MarlinFirmware/Marlin #25195: [BUG] Ender 3 V2 with DACAI LCD not working with Landscape

knuxyl, 2023-01-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I bought an Ender 3 V2 with RET6 chip that came with a DACAI LCD screen that I believe is the same one used in the Ender 3 S1. This screen only updates by using private folder. I set `define DWIN_MARLINUI_LANDSCAPE` and it is still portrait, altough it does look offset on the progress bar. I believe I tried setting portrait mode before and it was off as well. Creality UI works fine, although does not have landscape mode.

This was tested on 2.1.2, 2.1.X, and bugfix-2.1.X

Also, I don't know if it's actually called DACAI or what, I think I read that somewhere.

### Bug Timeline

Present since at least 2.1.2

### Expected behavior

Marlin UI to be in landscape mode

### Actual behavior

Marlin UI is in portrait mode with a lot of information cut off horizontally

### Steps to Reproduce

Flash firmware with `define DWIN_MARLINUI_LANDSCAPE` on Ender 3 V2 with newer DACAI screen.

### Version of Marlin Firmware

bugfix-2.1.X

### Printer model

Creality Ender 3 V2

### Electronics

Stock 4.2.2 RET6 board

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10356774/configuration.zip)

The-EG, 2023-01-06:
Take a picture of the back of the LCD board so it can be positively identified.

Unfortunately, it sounds like you have one of the clone LCDs that does not support the rotation function. If that is the case there's not going to be anything that can be done to get it to support landscape mode unless someone figures out how to get it into that mode. 
You should complain to Creality for holding a hardware lottery instead of selling consistent products.

ellensp, 2023-01-06:
Also hound Creality to release details on the display! It is complete undocumented.

vlad-the-compiler, 2023-01-06:
I have just received the same screen and can confirm your issues. It seems like the DACAI screen doesn't implement the same exact protocol.
Regarding text, right off the bat I can tell you that the DWIN variant uses bitmap fonts, whereas DACAI has a TTF renderer.
It is still unclear to me why the custom UIs work well but the simpler Marlin UI does not.

I'll do some investigation and check back with my findings.

vlad-the-compiler, 2023-01-07:
I've done some poking around and compiled a proper `private` bundle for DACAI screens that will work correctly with MarlinUI (_or at least initial testing suggests that it works_).

Please try to flash your screen with [https://github.com/vlad-the-compiler/e3v2-dacai-marlinui-test](this). It is worth noting that the repo I linked will not be up forever, as I intend to file a PR to the Marlin Configurations repo once I validate everything works correctly. Consider this a beta-test.

Also, I've found a bug in MarlinUI. It's an off-by-one glitch on the border for the X Y Z coordinates on the status screen. It only appears on DACAI screens as DWIN units seem to ignore out-of-bounds coordinates and rectify them to the display viewport. I'll open a PR for that as well, but for now, you may edit [https://github.com/MarlinFirmware/Marlin/blob/62507cf54b313e5c035e89263e2b2bf42dfc021d/Marlin/src/lcd/e3v2/marlinui/ui_status_480x272.cpp#L318](this line in ui_status_480x272.cpp) from `LCD_PIXEL_WIDTH` into `LCD_PIXEL_WIDTH - 1`.

2 additional notes:

- Display brightness seems to be on a non-linear curve for DACAI displays, as it only starts falling off under 50 (out of 255); I've checked with Creality's Ender 3 S1 repo (which uses this exact display) and there is no difference in protocol compared to what mainline Marlin adopted, so it appears this is a hardware difference
- Display rotation does not work indeed; I am not sure if this is due to the lack of the feature in the DACAI unit's firmware itself, or if the opcode is different. Again, the E3S1 repo uses the exact same protocol for rotating the display, yet the command fails on DACAI (but works on DWIN)

It appears that DACAI units are designed to clone the original DWIN controllers - and said cloning is incomplete. It might be that Creality initially adopted the DWIN displays when they wanted to implement a color display interface, but due to the parts shortages (or due to pricing?) they've also looked towards working with other contractors, and the easiest solution was to use display panels that are compatible _enough_ - by that I mean they've only cloned/adapted the functionalities that their firmware uses; yes, DWIN might've had a more robust feature set, but when they used modules from other manufacturers, they've only had them clone the required firmware features that their printers actually use - in this case leaving us with some pretty bad backlight control, and the lack of display rotation.

vlad-the-compiler, 2023-01-07:
> Also hound Creality to release details on the display! It is complete undocumented.

DWIN-based knob displays are documented. Not only are there tools to recompile asset bundles for them, but the API can be found in the Ender 3 V2 and S1 firmware repos. Marlin used that API in order to adapt the classic UI to those displays.

The undocumented LCDs are the newer S1 Plus and S1 Pro touch displays (which ironically are also in 2 variants, DWIN and DACAI) - we're still waiting on a firmware source code release for these, but you should bear in mind that these units are currently pretty dreadful, offerint _very_ limited functionality compared to what you can do with a knob display (be it the E3V2 / S1 color unit or the classic 128x64 displays).

ben-wallis, 2023-01-07:
> The undocumented LCDs are the newer S1 Plus and S1 Pro touch displays

Is there not another undocumented LCD, the one requiring the TJC_SET directory on the SD card to flash? I have one of these on a new Ender 3 S1 and as far as I can tell there's no Marlin firmware available for it that maintains the LCD functionality.

ellensp, 2023-01-07:
@ben-wallis there are currently 4 known LCD of that type that look the same on the outside.

The first was DWIN, well known, well documented..
the second was DACAI, an undocumented clone, but there are tools that converts from DWIN files to "private" files for this display..  (but it looks like it doesn't support some DWIN features)   
The third is VIEWE, another undocumented clone, no tools. 
The forth is TJC using TJC_SET, no documentation and no tools are available.

EvilGremlin, 2023-01-07:
What? They use TJC now? TJC have EVERYTHING available! It's probably best documented of them all (if you look at Nextion docs and forums)

vlad-the-compiler, 2023-01-07:
@ben-wallis  @ellensp I stand corrected, I did not know there are other compatible units other than DACAI.

Regarding VIEWE, from what I can see on the board, that looks like an STM32 clone - since the DWIN units are also seemingly based on STM32 (clones?), I wonder if those provide a 1:1 emulation of the original DWINs. Is there any firmware package available from Creality for those? I wouldn't be surprised if they used the same `DWIN_SET`.

And yes, as @EvilGremlin said, TJC have readily available tools, they're the manufacturers of the Nextion line of displays - they won't admit it, but disassembling their binaries and looking at their releases points to exactly that. Nextion is just the EU/US market-facing entity with a 200% markup.

closed 2023-01-15 by thinkyhead