MarlinFirmware/Marlin #25198: [BUG] #error "AUTOTEMP requires AUTOTEMP_MIN."

Dark360, 2023-01-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Attempted to compile Marlin bugfix-2.1.x using default Creality\Ender-3 Pro\CrealityV427 configuration using PlatformIO and Auto Build Marlin gives error #error "AUTOTEMP requires AUTOTEMP_MIN.".

Compiling Marlin 2.1.2 using default Creality\Ender-3 Pro\CrealityV427 configuration using PlatformIO and Auto Build Marlin works fine.

### Bug Timeline

New Bug 06/01/2023

### Expected behavior

I exxpected not to receive #error "AUTOTEMP requires AUTOTEMP_MIN." when compling.

### Actual behavior

Compile fails due to #error "AUTOTEMP requires AUTOTEMP_MIN.".

### Steps to Reproduce

1. Open Marlin-bugfix-2.1.x folder in Visual Studio Code
2. Copy Config files from Creality\Ender-3 Pro\CrealityV427 to Marlin folder
3. Compile using PlatformIO or Auto Build marlin

### Version of Marlin Firmware

Marlin-bugfix-2.1.x

### Printer model

Creailty Ender 3 Pro

### Electronics

4.2.7

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

cbagwell, 2023-01-06:
The compile failure is related to feature just committed and you likely were unlucky and grabbed a snapshot of Configuration before it was updated.  The needed values are in latest Configuration so I would update and recopy the file again.

Dark360, 2023-01-06:
> The compile failure is related to feature just committed and you likely were unlucky and grabbed a snapshot of Configuration before it was updated. The needed values are in latest Configuration so I would update and recopy the file again.

@cbagwell Which Configuration files should I be using with Marlin-bugfix-2.1.x?

https://github.com/MarlinFirmware/Configurations/archive/refs/tags/2.1.2.zip
or
https://github.com/MarlinFirmware/Configurations/archive/refs/heads/import-2.1.x.zip

Dark360, 2023-01-06:
I just answered my own question DUH!!

https://github.com/MarlinFirmware/Configurations/archive/refs/heads/bugfix-2.1.x.zip

Dark360, 2023-01-06:
I just answered my own question in my edit. 

https://github.com/MarlinFirmware/Configurations/archive/refs/heads/bugfix-2.1.x.zip

Thank you for your help @cbagwell @DrToxic

closed 2023-01-06 by Dark360

bsaverino, 2023-01-08:
This is still an issue. Using the most up to date branches as of 6 minutes ago I am getting the same error. @Dark360 what resolved this for you?

Using bugfix-2.1.x in both Marlin and Configuration.

ellensp, 2023-01-08:
@bsaverino look again.  

https://github.com/MarlinFirmware/Configurations/blob/import-2.1.x/config/examples/Creality/Ender-3%20Pro/CrealityV427/Configuration_adv.h#L453-L455