MarlinFirmware/Marlin #25200: [FR] add _USB Composite envir. to STM32G0

brew99, 2023-01-07:
### Is your feature request related to a problem? Please describe.

Unable to mount printer's SD card in Win10 laptop as there is no _USB environment to build.

### Are you looking for hardware support?

SKR mini E3 V3 board with STM32G0 MCU.  This is installed in an Ender 3

### Describe the feature you want

Be able to mount Printer's SD card in Win10 laptop, so that files can be directly transferred to the printer's SD card..  I see other BTT boards have the _USB envir.  Is this possible to add?

I previously used the Maple in the V1 board, but understand that Maple is now deprecated.

EvilGremlin, 2023-01-07:
It's cortex-m0, might be too simply slow to do composite with marlin...
Try setting same build flags/unflags, see how it goes.

brew99, 2023-01-08:
I tried adding an environment to the boards stm32g0xx.ini file, similar to how it is done for the stmf103, but it failed miserably on build.  Poked around for a few hours trying different things, but I just don’t have the knowledge to implement something like this.   Basically, I’m not a programmer 😄