MarlinFirmware/Marlin #25204: [BUG] In 2.1.2, Ender-2: 'LCD_PINS_RS' was not declared in this scope

probonopd, 2023-01-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Cannot compile Marlin 2.1.2 for Ender-2: 'LCD_PINS_RS' was not declared in this scope.

Marlin 2.1.1 works.

### Bug Timeline

2.1.2

### Expected behavior

Compiles.

### Actual behavior

Compilation fails.

### Version of Marlin Firmware

2.1.2

### Printer model

Creality Ender-2

### Electronics

Stock

### Add-ons

None

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

My configuration can be seen at https://github.com/probonopd/marlin-builds/blob/main/customization/configuration/Creality/Ender-2

ellensp, 2023-01-07:
https://github.com/MarlinFirmware/Marlin/commit/29202d5561cfd948c3224d2efe172f06bf5ae549   added 8 hours ago...

ellensp, 2023-01-07:
### real Steps to Reproduce:
Set   #define MOTHERBOARD BOARD_MELZI_CREALITY
and #define ENDER2_STOCKDISPLAY

You dont need all this other stuff complicating the issue

Bob-the-Kuhn, 2023-01-07:
What LCD are you using?  

There was a change made so that those pins are defined only when the **CR10_STOCKDISPLAY** is enabled.  This change was made in PR #25012.

If I set the LCD to ENDER2_STOCKDISPLAY then I get the errors you listed. **ENDER2_STOCKDISPLAY** is what is enabled in the Marlin example configuration for ENDER2.

As a short term work around you could make the following change to the **pins_MELZI_CREALITY.h** file:

1. delete line 48 `#if ENABLED(CR10_STOCKDISPLAY)`
2. add line 48 `#if EITHER(CR10_STOCKDISPLAY, ENDER2_STOCKDISPLAY)`

---

> My configuration can be seen at https://github.com/probonopd/marlin-builds/blob/main/customization/configuration/Creality/Ender-2

I'm confused.  That file has nothing to do with the standard Marlin.  Where are you getting your source files and config files?

probonopd, 2023-01-07:
I am using the stock display that came with the Ender-2.

ellensp, 2023-01-07:
@Bob-the-Kuhn he scripted github to build it for him,  using stock ender2 example conf, and then running his modify script to make configuration changes

probonopd, 2023-01-07:
Now I [get](https://github.com/probonopd/marlin-builds/actions/runs/3862628114/jobs/6584316260#step:10:159):

```
Compiling .pio/build/melzi_optimized/src/src/lcd/dogm/status_screen_DOGM.cpp.o
In file included from Marlin/src/lcd/dogm/marlinui_DOGM.cpp:42:0:
Marlin/src/lcd/dogm/marlinui_DOGM.cpp: In static member function 'static void MarlinUI::init_lcd()':
Marlin/src/lcd/dogm/marlinui_DOGM.h:231:34: error: 'DOGLCD_A0' was not declared in this scope
     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
                                  ^
Marlin/src/lcd/dogm/marlinui_DOGM.cpp:267:14: note: in expansion of macro 'U8G_PARAM'
     u8g.init(U8G_PARAM);
              ^~~~~~~~~
Marlin/src/lcd/dogm/marlinui_DOGM.h:231:34: note: suggested alternative: 'DOGLCD_CS'
     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
                                  ^
Marlin/src/lcd/dogm/marlinui_DOGM.cpp:267:14: note: in expansion of macro 'U8G_PARAM'
     u8g.init(U8G_PARAM);
              ^~~~~~~~~
*** [.pio/build/melzi_optimized/src/src/lcd/dogm/marlinui_DOGM.cpp.o] Error 1
```

Bob-the-Kuhn, 2023-01-07:
I screwed up.  Now that I saved the ENDER2_STOCKDISPLAY config I get the same errors.

Bob-the-Kuhn, 2023-01-07:
I just got around to compiling 2.1.1 with the default ENDER2 configs.  It compiled correctly.

There are two ways to get bugfix-2.1.x to compile:
1. Comment out line 52 in  pins_MELZI_CREALITY.h  `// #define LCD_PINS_DEFINED`
2. Restore the 2.1.1 version of line 142 in pins_SANGUINOLOLU_11.h:
- 2.1.1:  `#if HAS_WIRED_LCD`
-  bugfix-2.1.x:  `#if HAS_WIRED_LCD && DISABLED(LCD_PINS_DEFINED)`

Bob-the-Kuhn, 2023-01-07:
I just compiled it with **#define LCD_PINS_DEFINED** commented out and set the LCD to **CR10_STOCKDISPLAY**.  Now I get warnings that **LCD_PINS_RS** is being defined in two places.:

- **pins_MELZI_CREALITY.h** line 50
- **pins_SANGUINOLOLU_11.h** line167

Compiling with 2.1.1 and **CR10_STOCKDISPLAY** enabled.  No warnings.  Time to look further at the changes.

Bob-the-Kuhn, 2023-01-07:
Looks like the final fix is to keep LCD_PINS_DEFINED but add a couple more defines.  No more warnings or errors and M43 shows that the pin/function assignments are the same as with 2.1.1.
```
  #define DOGLCD_CS                           LCD_PINS_RS
  #define DOGLCD_A0                           LCD_PINS_D4
```

Please try this file and see if your system works.

[pins_MELZI_CREALITY.zip](https://github.com/MarlinFirmware/Marlin/files/10367224/pins_MELZI_CREALITY.zip)

probonopd, 2023-01-11:
Thank you very much @Bob-the-Kuhn. Compilation succeeded, will test the firmware on the machine asap.

ellensp, 2023-01-19:
@probonopd how goes that testing?

probonopd, 2023-01-20:
Thanks for the ping @ellensp, my test was successful :+1: 

I think this can be merged.

Bob-the-Kuhn, 2023-01-20:
PR #25258 fixes this bug so this issue is being closed.

closed 2023-01-20 by Bob-the-Kuhn