MarlinFirmware/Marlin #25205: [BUG] Auto homing hangs when input shaping is enabled

aleung, 2023-01-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Upgraded from 2.1.1 to latest 2.1.x or bugfix-2.1.x and enable the input shaping feature. When auto homing, the machine hangs.

I disabled input shaping feature and re-fresh the firmware, the auto homing worked fine.

### Bug Timeline

Since 2.1.2

### Expected behavior

I expected it to homing successfully.

Below is the debug output of a successful homing when I disable input shaping feature:

```
G28
Ok
>>> G28  X0.00 Y0.00 Z235.00
Machine Type: Delta
Probe: FIX_MOUNTED_PROBE
Probe Offset X0.00 Y0.00 Z-0.30 (Aligned With & Below Nozzle)
>>> home_delta  X0.00 Y0.00 Z235.00
  current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
>>> homeaxis(X)
Home Fast: 352.50mm
>>> do_homing_move  X0.00 Y0.00 Z246.41
...(X, 352.50, [50.00])
<<< do_homing_move  X0.00 Y0.00 Z246.41
Move Away: -5.00mm
>>> do_homing_move  X0.00 Y0.00 Z246.41
...(X, -5.00, [50.00])
<<< do_homing_move  X0.00 Y0.00 Z246.41
Re-bump: 10.00mm
>>> do_homing_move  X0.00 Y0.00 Z246.41
...(X, 10.00, 5.00)
<<< do_homing_move  X0.00 Y0.00 Z246.41
adjDistance:-0.30
>>> do_homing_move  X0.00 Y0.00 Z246.41
...(X, -0.30, 5.00)
<<< do_homing_move  X0.00 Y0.00 Z246.41
<<< homeaxis(X)
>>> homeaxis(Y)
Home Fast: 352.50mm
>>> do_homing_move  X0.00 Y0.00 Z246.41
...(Y, 352.50, [50.00])
<<< do_homing_move  X0.00 Y0.00 Z246.41
Move Away: -5.00mm
>>> do_homing_move  X0.00 Y0.00 Z246.41
...(Y, -5.00, [50.00])
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z246.41
Re-bump: 10.00mm
>>> do_homing_move  X0.00 Y0.00 Z246.41
...(Y, 10.00, 5.00)
<<< do_homing_move  X0.00 Y0.00 Z246.41
adjDistance:-0.99
>>> do_homing_move  X0.00 Y0.00 Z246.41
...(Y, -0.99, 5.00)
<<< do_homing_move  X0.00 Y0.00 Z246.41
<<< homeaxis(Y)
>>> homeaxis(Z)
Home Fast: 352.50mm
>>> do_homing_move  X0.00 Y0.00 Z246.41
...(Z, 352.50, [50.00])
<<< do_homing_move  X0.00 Y0.00 Z246.41
Move Away: -5.00mm
>>> do_homing_move  X0.00 Y0.00 Z246.41
...(Z, -5.00, [50.00])
<<< do_homing_move  X0.00 Y0.00 Z246.41
Re-bump: 10.00mm
>>> do_homing_move  X0.00 Y0.00 Z246.41
...(Z, 10.00, 5.00)
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z246.41
<<< homeaxis(Z)
>>> set_axis_is_at_home(X)
  current_position= X0.00 Y0.00 Z246.41 : 
<<< set_axis_is_at_home(X)
>>> set_axis_is_at_home(Y)
  current_position= X0.00 Y0.00 Z246.41 : 
<<< set_axis_is_at_home(Y)
>>> set_axis_is_at_home(Z)
  current_position= X0.00 Y0.00 Z236.41 : 
<<< set_axis_is_at_home(Z)
  current_position= X0.00 Y0.00 Z236.41 : sync_plan_position
<<< home_delta  X0.00 Y0.00 Z236.41
>>> do_blocking_move_to  X0.00 Y0.00 Z236.41
>  X0.00 Y0.00 Z178.59
  destination= X0.00 Y0.00 Z236.41 : destination = current_position
  destination= X0.00 Y0.00 Z178.59 : prepare_fast_move_to_destination
  current_position= X0.00 Y0.00 Z178.59 : zone border move
  current_position= X0.00 Y0.00 Z178.59 : xy move
<<< do_blocking_move_to  X0.00 Y0.00 Z178.59
X:0.00 Y:0.00 Z:178.59 E:0.00 Count A:36992B:36992C:36992
<<< G28  X0.00 Y0.00 Z178.59
ok
```

### Actual behavior

```
G28
Ok
>>> G28  X0.00 Y0.00 Z235.00
Machine Type: Delta
Probe: FIX_MOUNTED_PROBE
Probe Offset X0.00 Y0.00 Z-0.10 (Aligned With & Below Nozzle)
>>> home_delta  X0.00 Y0.00 Z235.00
  current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
>>> homeaxis(X)
Home Fast: 352.50mm
>>> do_homing_move  X0.00 Y0.00 Z246.21
...(X, 352.50, [50.00])
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z246.21
Move Away: -5.00mm
>>> do_homing_move  X0.00 Y0.00 Z246.21
...(X, -5.00, [50.00])
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
```

The printer stopped moving. Kept repeating "busy: processing" output and didn't response to any command.

### Steps to Reproduce

1. Fresh the firmware with `INPUT_SHAPING_X` and `INPUT_SHAPING_Y` enabled
2. Perform auto homing by operating on controller or send G28 via console.

### Version of Marlin Firmware

2.1.2

### Printer model

Kossel Mini (Delta)

### Electronics

MKS TinyBee

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10366876/Configuration.zip)

thinkyhead, 2023-01-08:
Thanks for the report! We're working to figure out the reason for these hangs and prevent them, if possible. Meanwhile, we may just add code to disable input shaping for certain internal moves initiated by commands like `G28` and `G29`.

Mjankor, 2023-01-08:
And perhaps g27 too.

thinkyhead, 2023-01-12:
For now we're going to forbid the use of Input Shaping with `DELTA`. Later on we may implement the shaper in a way that supports kinematic systems, but it's unclear whether `DELTA` can ever be supported.

closed 2023-01-12 by thinkyhead

HunterLYZ, 2023-02-11:
Same mks tiny bee, same behavier, but Cartesian 3d printer; Have you tried linear advance, It causes the extruder not working for me. I guess there is a connection with
[#25160](https://github.com/MarlinFirmware/Marlin/issues/25160#issue-1513116272)
Both are related to I2S motor stream.

brdvvoz, 2023-02-13:
Same thing on MKS Robin Nano 3.1
Problem seems to disappear when lowering microsteps on X and Y (128->64 in my case)