MarlinFirmware/Marlin #25209: LK4 PRO

HajmamadAgha, 2023-01-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I got the LK4 PRO Board with DMT48270C043_06WT touchscreen LCD.
I honestly have no idea which branch  to use? (LKxPro-Marlin1.1.9-master, Longer-LK4-PRO-main, Marlin2.0-lgt-support-LKxPro, Marlin-2.1.2)
I can monitor LGT_KIT board uploading Marlin branches but nothing shows up on display

### Bug Timeline

All version

### Expected behavior

I expected it to work.

### Actual behavior

Uploaded the specific Longer-LK4-PRO-main on board.
Can monitor stuff.
Nothing shows up on LCD.
Crying

### Version of Marlin Firmware

Marlin-2.1.2

### Printer model

LK4 PRO

### Add-ons

DMT48270C043_06WT

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

https://drive.google.com/file/d/1A2KLzKHLpUmx2S6jvm7srz09vj3QxTwG/view?usp=sharing

Bob-the-Kuhn, 2023-01-08:
I hate to point you away from mainstream Marlin but ... here's a link to a [site](https://github.com/LONGER3D/Marlin2.0-lgt/tree/support-LKxPro?fbclid=IwAR3A7qlO3rM-8X-T8iFUw3-0RMu9QKQbRXN7pyFKP5Fewsikg-DhdpGHkG4) that appears to support your system.  I don't know if it supports your display.

You could use the info on that site to create mainstream Marlin firmware.  The easiest way is to compare the config files on that site with the ones in [Marlin's LK5 PRO example](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Longer/LK5) and transfer the LK4 PRO specific info over.  

Your display could be a problem.  I believe the correct Marlin setting would be to enable DWIN_MARLINUI_PORTRAIT.  I think the biggest issue will be defining the interface.  My understanding is that it's a simple serial link.  What we'd need to determine is what the settings are for LCD_SERIAL_PORT and LCD_BAUDRATE.  These need to be added to configuration.h:
```
#define LCD_SERIAL_PORT ?
#define LCD_BAUDRATE 115200   // :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000] Enable to override BAUDRATE
```

Please verify what motherboard you have.

ellensp, 2023-01-08:
"LKxPro-Marlin1.1.9-master, Longer-LK4-PRO-main, Marlin2.0-lgt-support-LKxPro" none of those are part of Marlin, they are all in forks.  (A fork is a Marlin added to and supported and maintained by a 3rd party)  

If the source is not on https://github.com/MarlinFirmware/  you cannot get support for it here

Official marlin only has support for the LK5, because this use a stock 128x64 display. 

The stock lcd on the  LK4 PRO is not supported in stock Marlin.

Bob-the-Kuhn, 2023-01-08:
@ellensp 

The DMT48270C043_06WT is sometimes referred to as a DWIN.  Is there a Marlin setting for it?  I was hoping DWIN_MARLINUI_PORTRAIT was the correct one.

I'm working on a facebook forum with a person that has this display.  It would be nice to know for sure if it is supported by mainstream Marlin.

Bob-the-Kuhn, 2023-01-08:
I've just put my toe into the DWIN/DGUS pool.  Mighty deep.

Sounds like we'd need to reprogram the DMT48270C043_06WT with DGUS software and then enable DGUS_LCD_UI ORIGIN.  Doesn't sound like fun.

ellensp, 2023-01-08:
yes that is the issue...  it runs it own unique UI, marlin doesn't know how to talk to the vendor supplied UI.

Yes the display is a stock T5UID1 DWIN

We have 3 different UI for it
ORIGIN
T5UID1
and even IA_CREALITY  

But all of these are Horizontal UI, not Vertical UI like this printer comes with

HajmamadAgha, 2023-01-09:
I'm a noob to this. so the motherboard is LGT_KIT_V1.0 (LK4 PRO with mega2560 processor).
I used the configurations and etc on this website [](https://github.com/ils15/Longer-LK4-PRO).
If I've got it right I should test all LCD_SERIAL_PORTs and LCD_BAUDRATEs to make things right and at the same time reprogram DMT48...WT via DGUS software. Right?

Bob-the-Kuhn, 2023-01-10:
You may want to change MOTHERBOARD BOARD_RAMPS_13_EFB to BOARD_LONGER3D_LKx_PRO.  I believe that the LK5 PRO and the LK4 PRO use the same motherboard.  The LONGER3D_LKx_PRO motherboard wasn't available in the older version of Marlin used by https://github.com/ils15/Longer-LK4-PRO.

You'll also need to verify what stepper drivers are on your motherboard.  I've seen A4988, TMC2208_STANDALONE and TMC2209_STANDALONE specified.  Sometimes they are mixed on the board.

HajmamadAgha, 2023-01-14:
Okay, So I changed everything on configuration.h and configuration_adv.h.
But about reprogramming the LCD, how should it be done?

rick090980, 2023-02-04:
Being that I currently own a LK4 Pro I can confidently say your only feasible solution is to swap in an aftermarket motherboard and display to something more suited to Marlin 2.0 and up. The LK4 Pro board (at least the one that came with mine) is an 8 bit board, as opposed to 32 bit, and was never designed to run Marlin 2. I personally use a BTT SKR 2 board with TMC2209 drivers in UART, and their TFT35 E3 dual mode LCD. I couldn't be happier.