MarlinFirmware/Marlin #25210: [BUG] MKS SGEN L V1.0 - "Serial port pins (3) conflict with Z2 pins!"

Papadoma, 2023-01-08:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Firmware won't compile showing the error:
`In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/../HAL/LPC1768/inc/SanityCheck.h:176:6: error: #error "Serial port pins (3) conflict with Z2 pins!"`

Previous firmware was Marlin 2.0.9.3, which compiles fine. Transferred changes from old to new config files.

I am using the E1 driver as a Z2 driver, to have auto gantry leveling. All drivers are TMC2209.

### Expected behavior

Successful compilation

### Version of Marlin Firmware

Marlin 2.1.2

### Electronics

MKS SGEN L V1.0

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10368557/Configuration.zip)

ellensp, 2023-01-08:
Sigh, you add a sanity check to stop users making mistakes so they report it as a bug..

You board has these pins 
```
#define E1_STEP_PIN             P0_01     //  used for Z2
#define E1_DIR_PIN                P0_00       //  used for Z2

```
But 

```
Serial3 TX P0_00  
Serial3 RX P0_01

```
There are obviously in conflict

Where is serial port 3 coming from 

You have a serial display DGUS_LCD_UI_MKS
You don't have LCD_SERIAL_PORT it is auto assigned to port 3

So this configuration is in error, as the error message says

I dont know how you physically have this connected. two easy options

using exp1  the only serial port I see on there is serial port 1
So that would  require #define LCD_SERIAL_PORT 1 in your configuration.h

using aux-1 This has serial port 0 on it.
In wich case you need #define LCD_SERIAL_PORT 0 and to disable #define SERIAL_PORT_2 0

closed 2023-01-08 by ellensp

Papadoma, 2023-01-08:
> Sigh, you add a sanity check to stop users making mistakes so they report it as a bug..
> 

Being patronising surely does not help anyone.
Since the pins didn't change between the two versions and this LCD_SERIAL_PORT detail is not mentioned in comments in the code, then yes it is in fact a bug.

Adding #define SERIAL_PORT_2 0 allow compilation to proceed, only to fail with a new error of 
`Marlin\src\gcode\bedlevel\abl\G29.cpp:439:33: error: 'onLevelingStart' is not a member of 'ExtUI'; did you mean 'onHomingStart'?`
Changing bed levelling from linear to bilinear allows the compilation to finish successfully. Is that another case of a user mistake or could it be a bug?