MarlinFirmware/Marlin #25213: Error "Z_PROBE_SPEED_FAST is now Z_PROBE_FEEDRATE_FAST.\"

GRAZIANO2209, 2023-01-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

resource": "/c:/Users/Notebook/Downloads/Marlin-2.1.x (1)/Marlin-2.1.x/Marlin/src/inc/SanityCheck.h",
	"owner": "cpp",
	"severity": 8,
	"message": "#error \"Z_PROBE_SPEED_FAST is now Z_PROBE_FEEDRATE_FAST.\"",
	"startLineNumber": 147,
	"startColumn": 4,
	"endLineNumber": 147,
	"endColumn": 4
},{
	"resource": "/c:/Users/Notebook/Downloads/Marlin-2.1.x (1)/Marlin-2.1.x/Marlin/src/module/motion.h",
	"owner": "cpp",
	"severity": 8,
	"message": "'Z_PROBE_FEEDRATE_FAST' was not declared in this scope; did you mean 'Z_PROBE_FEEDRATE_SLOW'?",
	"startLineNumber": 69,
	"startColumn": 55,
	"endLineNumber": 69,
	"endColumn": 55
}]

### Bug Timeline

New problem started when I run firmware compilation

### Expected behavior

I expected you to compile

### Version of Marlin Firmware

2.1.x. bugfix

### Printer model

DELTA Printer

### Electronics

BTT SKR PRO 1.1

### Add-ons

BLTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-01-08:
> `#error "Z_PROBE_SPEED_FAST is now Z_PROBE_FEEDRATE_FAST.`

You need to update your config and replace `Z_PROBE_SPEED_FAST` with `Z_PROBE_FEEDRATE_FAST`.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-01-08 by thisiskeithb

GRAZIANO2209, 2023-01-08:
I did the configuration ex new reporting only the parameters of the mechanics of the printer, I do not understand where the error comes from.

thisiskeithb, 2023-01-08:
You have `#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z` on line 1506 in `Configuration.h`.

This isn't a support forum, so please use the links above for further help with your configuration.