MarlinFirmware/Marlin #25215: [FR] (feature summary) Fan Invert (Usefully for PWM Fans)

Mordi1984, 2023-01-09:
### Is your feature request related to a problem? Please describe.

Fan Inversion for PWM Fans.

### Describe the feature you want

I have an Octopus Pro F429 and want to controll one Fan (24V Papst) with PWM for my Chamber Heater. If i connect it to my "CNC-FAN" aka PWM Fan it spins a way to much and very loud (almost 90db).
If i Set the the Fan to 100% or 255 the Fan stops.
I would only use an Fan Range to 40% at much.
It is the only one PWM Fan i have in my Setup. all other are regular Fans.

Kind Regards.

Mordi

thisiskeithb, 2023-01-09:
> I would only use an Fan Range to 40% at much.

This exists as `FAN_MAX_PWM`. A value of 102 would be 40%

https://github.com/MarlinFirmware/Marlin/blob/202114c018dc22513e74c5b76e92ecda53214310/Marlin/Configuration_adv.h#L573-L586

closed 2023-01-09 by thisiskeithb

Mordi1984, 2023-01-09:
Hwllo Keith.
Thank you for your help.
I habe a Question. 255 is in my case 0 Fan speed and 153 would be 40% and 0 is full speed.
Can i reverse that settings?
What do my other Fans do! Will they also get inverted?
They run also on PWM Fan pins.

Maybe that question is wird, but i need to understand this.

Kind regards,
Mordi

Closed #25215<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FMarlinFirmware%2FMarlin%2Fissues%2F25215&data=05%7C01%7C%7C875dc9e3e0af49556c6c08daf1ea744a%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638088286512210982%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=d9eA5rXwga%2FvcmkRlE86nf76rcc4FZUlq5gA2MOO5lA%3D&reserved=0> as completed.

thisiskeithb, 2023-01-09:
Add this to your config to invert fan PWM:

`#define FAN_INVERTING`

Mordi1984, 2023-01-09:
> Is #define FAN_INVERTING_5 also possible to invert only one fan index 5?
>
> Not currently. Similar to FAN_MIN/MAX_PWM, all fans are configured by a single FAN_INVERTING setting.

Well, also it would be a feature we may need. Or i need to reverse that with an Hardware solution. Or i need to change all fans.

What do you think, is there an option that it may get programmed to prevent Hardware modifications?

Kind regards Keith and i want to thank you for your Help.