MarlinFirmware/Marlin #25217: M1 not recovering after blocking HOST

inspectionsbybob, 2023-01-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Occasionally, when running Repetier Host (like OctoPrint) Marlin FREEZES.  Sometimes I will see a "Printer Blocking" message (M1) come up on my Repetier console, but often nothing.  Touching my printer from Panel "un-freezes" the printer and it continues on where it left off.

This SEEMS to happen most on small parts with LOTS of small moves... ie Buffer full...

The issue is that it doesn't recover from this, or send the "Host" a follow-up message of "I'm ready again" type message.,

### Bug Timeline

Gotten worse with latest bugfix release (perception anyway)

### Expected behavior

I don't expect it to get "Stuck"

No such problem when running from SD card... only from a "HOST"

### Version of Marlin Firmware

Current Bugfix

### Printer model

JGMaker Artist

### Bed Leveling

None

### Your Slicer

Other (explain below)

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

So I can look at it, where is the M1 triggered from, is there SUPPOSED to be follow-up command, and where in the code should I look?

Is there an RTS/CTS equivalent on the USB connection I should look at on the HOST side?

thisiskeithb, 2023-01-09:
> Don't forget to include
> A ZIP file containing your Configuration.h and Configuration_adv.h.

You forgot to do this.

inspectionsbybob, 2023-01-10:
Sorry, forgot the Config files.

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/10385523/Configuration_adv.zip)

inspectionsbybob, 2023-03-12:
Most of the issues were related to a seriously overloaded Host CPU.  CPU percentages were often over 87% and it was causing a host of problems such as Nits on prints, missing commands, and other things.  

When you have a REALLY weird issue, check the Host spooler to make sure it is not having problems.
Octoprint and Repetier are very very stable but can (ALSO) have problems when they are running short on cycles.

closed 2023-03-12 by inspectionsbybob