MarlinFirmware/Marlin #25220: [BUG] SENSORLESS BACKOFF is not moving in the right direction.

DimchikUA, 2023-01-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When home direction is set to Xmax Ymax, _SENSORLESS_HOMING_, _SENSORLESS_BACKOFF_MM_ and _QUICK_HOME_ are enabled, BACKOFF moves towards Xmax Ymin. Although it should be також Xmin.
Because of this, the belt is heavily stressed, and the motor loses steps when the carriage is already at the end of the path, and I try to execute G28.
When _QUICK_HOME_ is turned off, everything works fine.

### Bug Timeline

My previous firmware version 2.0.9.1 also had this problem.

### Actual behavior

I expect SENSORLESS_BACKOFF to move in the opposite direction from home_direction, but this only works for the Y-axis. The X-axis moves in the same direction as home_direction, which is Xmax.

### Steps to Reproduce

G28

### Version of Marlin Firmware

Marlin 2.1.2

### Printer model

my coreXY build

### Electronics

SKR1.3

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10382748/Marlin.zip)

Bob-the-Kuhn, 2023-01-13:
I don't understand what is going on.  Perhaps enabling the debug prints will give us some more info.

Please enable **MARLIN_DEBUG_LEVELING** in **configuration.h** and reflash the motherboard.

After each powerup issue the **M111  S32** command.  That'll enable the debug prints.  Now Issue G28 and log the results.

ellensp, 2023-01-20:
Trying to test this config, I cannot as #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."

And sure enough #define E0_AUTO_FAN_PIN P2_03 while   
#define FAN_PIN                          P2_03

Not that this is in any way related, but indicates you may have changes pins

DimchikUA, 2023-01-22:
Yes, I did change some of the pins to make it easier to connect the fan.  I think that it does not affect where it will be directed backoff.
Unfortunately, there is no way to do tests with MARLIN_DEBUG_LEVELING. Maybe a little later.
But I am sure that I have described the problem in sufficient detail, and it is clearly visible in the attached video. The first small movement along the axis X is backoff, and it occurs in the co-direction with home direction, not in the opposite direction.

zalexzperez, 2023-01-31:
> I don't understand what is going on.

https://youtube.com/shorts/veJpc6tv0E8

When Auto homing with Quick Home enabled, the X axis first moves to the origin a certain distance, and THEN backs off. That's what causes the issues (grinding against the frame, missed steps, if the toolhead was near the origin before auto homing)

That initial movement from quick home needs to happen AFTER the sensorless backoff, not prior to it.

closed 2023-04-13 by github-actions[bot]