MarlinFirmware/Marlin #25222: Auto homing fails to take z-probe offset into account

jwalton, 2023-01-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I ran the z-probe wizard, which measured my z-probe offset as -1.8mm (1.8mm below the nozzle height).

If I now auto-home the printer, the bed comes up and touches the probe twice, then descends to 10mm.  If I raise the bed up to 1.8mm, the nozzle is touching the bed.  If I move it all the way up to 0, the nozzle collides with the bed.

### Expected behavior

Auto home should zero the bed to the height of the nozzle, not the height of the nozzle minus the z-probe offset.

### Actual behavior

Auto home sets the 0-point of the bed to be `z-probe offset` too low.

### Steps to Reproduce

1. Auto home printer

### Version of Marlin Firmware

bugfix-2.1.x (33e5aad364)

### Printer model

Creality Ender 5 Pro

### Electronics

4.2.2 silent mainboard

### Add-ons

CRTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/10383752/Config.zip)

AustinTylerDean, 2023-01-12:
I have a similar issue, though may be different.
I use the Z Offset Wizard, determine that the delta is -1.9mm. Click Done, then Save.
Use the wizard AGAIN, find a delta of -.25, Done, then Save.
Z Offset value now shows -.25 instead of -2.15, resulting in having to zero out your SW Z Offset every time before you use the wizard.
Hoping these are the same things, it would certainly be a huge fix in my book.

NeoIsrafil, 2023-01-15:
Honestly, i'm thinking nozzle to probe offset is badly broken.  Certain things take it into account, certain things dont, it may have to do with the change to  the way marlin handles the z clearance deploy probe, z clearance between probes, etc.  it may also have to do with something entirely different.  I'm just speculating.  I do know that when i have anything in nozzle to probe offset for z it does weird things, the z clearance also does weird things, its needing a z probe low point of -4 to even trigger the second time it tries to touch off, because it jumps up 30mm or so after the first and i have no idea where its getting that command from.

Has anyone found a workaround?  I'd like my printer working :(

dynek, 2023-01-17:
@AustinTylerDean I'd need to check it again but I have to zero out the value before running the z-offset probe wizard, because it will actually **add** delta each time.

dynek, 2023-01-18:
@AustinTylerDean your reasoning is correct. From a very simple perspective, Z-Offset should be set to the value discovered during calibration, nope ?

Just made a test:

```
Send: M851 Z0
Recv: ok
Send: M851
Recv:   M851 X45.00 Y8.00 Z0.00 ; (mm)
```

Running Z Probe Wizard that ends up displaying:
> Z-0.80
> Probe Z Offset-0.80

Saving data, M851 now shows Z-0.80 👍 
But if I now re-run the Z Probe Wizard, I will find out the same value (-0.8) but it will add it to previous value.
It ends up displaying:
> Z-0.80
> Probe Z Offset-1.60

And value stored is -1.60 and now the nozzle hits the bed.

mashuptwice, 2023-01-19:
The same seems to happen with babystepping. I have an exponentially growing error in the offset, started at -1.7mm and now I am at +7.490mm!

vito30x, 2023-01-20:
> Honestly, i'm thinking nozzle to probe offset is badly broken. Certain things take it into account, certain things dont, it may have to do with the change to the way marlin handles the z clearance [...]

Got the same Problem here. This happen after updating to Marlin 02010300.

vito30x, 2023-02-18:
Is there any fix done yet?

inspectionsbybob, 2023-03-13:
I am seeing the same thing that my Z-offset keeps growing after every reset.

Everything is stable and works for days and days.  But if I have to reset the printer due to a failed print, or something else, the Z-offset ALWAYS is off, and needs to be made larger (more negative).

closed 2023-06-02 by github-actions[bot]