MarlinFirmware/Marlin #25229: [BUG] X2 point for safe homing not working

polamoros, 2023-01-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm configuring Marlin for a custom Sigma R19 with an Octopus motherboard (from BigTreeTech).

I configured the second carriage using the DUAL_X_CARRIAGE.

When homing, I need to move the second carriage out of bed, but using Z_SAFE_HOMING_X2_POINT doesn't work.

### Bug Timeline

New bug

### Expected behavior

I expect the second carriage (X2) to move to the desired position.

### Actual behavior

It does nothing.

### Version of Marlin Firmware

2.1.x

### Printer model

Custom Sigma r19

### Electronics

Octopus Motherboard

### Bed Leveling

ABL 3-point

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/10391746/Archive.zip)

thisiskeithb, 2023-01-11:
> using Z_SAFE_HOMING_X2_POINT doesn't work.

Because that isn’t a valid define.

thisiskeithb, 2023-01-11:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/16577

closed 2023-01-11 by thisiskeithb

polamoros, 2023-01-12:
Thanks for the information! 

Unfortunately, the issue you linked was closed without any solution.
.
If there is no option to set safe Homing positions for the 2nd tool on IDEX systems, do you know any other approach to achieve the same fix?

I really appreciate any help you can provide.

thisiskeithb, 2023-01-12:
> Unfortunately, the issue you linked was closed without any solution.

You might want to check that again. It's still an open feature request:

[image]

> do you know any other approach to achieve the same fix?

I do not. You'll have to wait for the linked feature request to be implemented.

polamoros, 2023-01-12:
Thank you so much for your answer!