MarlinFirmware/Marlin #25235: [BUG] (bug summary)

FlailAway, 2023-01-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Navigate "Config > advanced > temp > AutoTemp
Change "Off" to "on"
Change Max 250 to "260"
Return
Return
Save Settings - it beeps and shows "Settings saved"
Do print
Turn off
Next time turn on, AutoTemp is OFF and Max is 250

### Bug Timeline

Only just tried using autotemp.

### Expected behavior

Stay "ON" after a restart.

### Actual behavior

As above.

### Steps to Reproduce

As above

### Version of Marlin Firmware

2.2.x bugfix shows 2010200 in config

### Printer model

Home built corexy

### Electronics

MKS-GenL

### Add-ons

none

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

FlailAway, 2023-01-12:
Here is .zip as it did not give me an option to attach it with the OP. 
[bug-no-save.zip](https://github.com/MarlinFirmware/Marlin/files/10405766/bug-no-save.zip)

thisiskeithb, 2023-01-12:
Saving AUTOTEMP to EEPROM was added two weeks ago: https://github.com/MarlinFirmware/Marlin/pull/25093

Your config is out of date and does not include these options (and is missing a lot of other updates), so you are not testing the latest `bugfix-2.1.x`.

[image]

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-01-12 by thisiskeithb