MarlinFirmware/Marlin #25237: [FR] (Separate steps/mm settings for second carriage for IDEX)

TheThomasD, 2023-01-13:
### Describe the feature you want

I have an IDEX printer and of course there are two steppers involved to move the independent heads. I looked through the code but couldn't find any hint on how to set the steps/mm for the second stepper motor (X2).

I'd like to be able to set the steps/mm separately for the second stepper to be able to have different settings for X and X2 axes.

gsuresh2u, 2023-01-16:
Hi, i need that feature too
and also wanted X2 motor reverse direction in Marlin, presently there is no opttion in Marlin to revert X2 motor direcion in IDEX mode

EvilGremlin, 2023-01-16:
Doubt this will happen, it calls for duplication/mirror mode rewrite, big complication for duibous edge case benefit. The only reason to set different steps/mm is different mechanics, like 16 and 20 teeth gears, which is bad design in the first place

inspectionsbybob, 2023-03-30:
Both of those features are ALREADY IN MARLIN.
I have an IDEX printer, and I have different Estep setting for both extruders.
There are also setting for reversing each/every stepper on a stepper-by-stepper basis.  Benn using them since early 2.0.x
I know because my second extruder is wired backwards (thanks factory) so I have to use the setting.

ellensp, 2023-03-30:
@inspectionsbybob 
For E, yes but they want different X steps for X and X2

ellensp, 2023-03-30:
could be mixed steeper motors ie 1.8 degree and a 0.9 (yea madness, I agree) 
Or a machine for batch printing where X2 is on a different part of the machine with different hardware.

LittleNyanCat, 2023-07-25:
You can simply reverse the connector. If you have JST XH connectors, you can remove each wire individually from the connector with a thumbtack and then put them in reverse order

> Hi, i need that feature too and also wanted X2 motor reverse direction in Marlin, presently there is no opttion in Marlin to revert X2 motor direcion in IDEX mode

muttstang, 2024-03-08:
> What kind of printer has multiple Esteps on the X-Axis? I can't envision that one. Is there a totally separate X-Axis carriage for the second (or more) extruder?

It can be a belt tension issue, the X2 carriage belt may not be at exactly the same tension as the X1 so the steps/mm would end up being different to get a dimensionally accurate print.  What I've resorted to is adjusting tension on one of the belts until they are the same steps/mm