MarlinFirmware/Marlin #25241: [BUG] Anet ET5X Screen Colors/Antialiasing?

dady172172, 2023-01-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I installed bugfix-2.1.x on my ET5X and noticed the screen looks different from 2.1.x and 2.0. Mostly the colors seem to be off and some kind of shadow affect for letters and numbers, which is not really a shadow. More of a Blue pixle mess. I have always used the default configs for each version (Downloaded from the file in the config folder for each branch).  I have talked with thisiskeithb, who has confirmed the screen issue.. Thank you thisiskeithb for the hard work on the configs they work great..

Normal photo of 2.1x
[image]

Weird colors shadows Bugfix-2.1.x
[image]

Configs:
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/10416768/Configs.zip)

### Bug Timeline

Bugfix-2.1.x

### Expected behavior

For the screen to not be so hard to look at. If you look at it long enough it kinda hurts your eyes a little bit.

### Steps to Reproduce

Update bugfix-2.1.x on ET5X

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Anet ET5X

### Electronics

Stock

### Add-ons

None

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-01-14:
I can confirm this on the same Anet ET5 / `ANET_ET5_TFT35`. Interestingly, the `ANET_ET4_TFT28` TFT looks fine.

I also tested TFT_COLOR_UI on some other SPI screens and don't get these odd colors, but I have a couple more TFT types to test this weekend.

dady172172, 2023-01-19:
Changing `Line 3230 in Configuration.h` ` #define TFT_FONT  NOTOSANS` to `#define TFT_FONT  HELVETICA` fixes the colors and antialising issue. Looks like it did in 2.0.
[image]

thisiskeithb, 2023-01-19:
@jmz52 might have some ideas.

jmz52, 2023-01-20:
@thisiskeithb, can you tell me how `ANET_ET4_TFT28` is connected to MB - SPI, FSMC16 or FSMC8 like `ANET_ET5_TFT35`?

ellensp, 2023-01-20:
```
#elif EITHER(LERDGE_TFT35, ANET_ET5_TFT35)                                    // ST7796
  #define TFT_DEFAULT_ORIENTATION TFT_EXCHANGE_XY
  #define TFT_RES_480x320
  #define TFT_INTERFACE_FSMC
#elif ANY(ANET_ET4_TFT28, MKS_ROBIN_TFT24, MKS_ROBIN_TFT28, MKS_ROBIN_TFT32)  // ST7789
  #define TFT_DEFAULT_ORIENTATION (TFT_EXCHANGE_XY | TFT_INVERT_Y)
  #define TFT_RES_320x240
  #define TFT_INTERFACE_FSMC

```

jmz52, 2023-01-20:
Yeah, I've saw this in pins file.
The problem is if both displays are with the same 8-bit FSMC interface, I can't explain why only one of them is affected by _blue pixie's mess_.
So I was hoping there is something different with `ANET_ET4_TFT28`

davidtgbe, 2023-01-20:
> I can confirm this on the same Anet ET5 / `ANET_ET5_TFT35`. Interestingly, the `ANET_ET4_TFT28` TFT looks fine.
> 
> I also tested TFT_COLOR_UI on some other SPI screens and don't have these odd colors, but I have a couple more TFT types to test this weekend.

This is happening to me also in the ET4 with NOTOSANS (AA) font.

dady172172, 2023-01-20:
> This is happening to me also in the ET4 with NOTOSANS (AA) font.

Did changing `TFT_FONT  HELVETICA` work for you?

davidtgbe, 2023-01-21:
> Did changing `TFT_FONT HELVETICA` work for you?

Yes it works fine with Helvética font. Maybe there is a problem with the antialiasing color transition?

jmz52, 2023-01-23:
Should be fixed in https://github.com/jmz52/Marlin/tree/blue-pixie's-mess but I do not have a board with 8-bit FSMC interface to test it myself.

dady172172, 2023-01-25:
compiled blue-pixie's-mess with default config and screen is all good now..
[image]
[image]

thisiskeithb, 2023-01-25:
Closing since a PR has been opened:

- https://github.com/MarlinFirmware/Marlin/pull/25285

closed 2023-01-25 by thisiskeithb