MarlinFirmware/Marlin #25242: [BUG] "Filament Change" menu selection gets stuck "waiting for user input" on Ender 3 V2/V2 Neo with DWIN/DACAI screens + Jyers UI

ngth82, 2023-01-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Originally i had an issue with M600 not working on Marlin 2.1.2 and a Ender 3 V2 (DWIN screen) and a V2 Neo (DACAI screen). Both are running JyersUI. M600 was resolved with the notes from issue #24996 .

However, in the Prepare Menu, there is also an option to "Change Filament" - the hot end correctly heats up, and unloads and loads up again with a purge, but then it gets stuck "Waiting for user input". There is no way around this. 

I tried looking at the serial terminal - it only spits out busy messages. 

Attempts at issuing an M108 (Break and Continue) doesn't do anything either. 

### Bug Timeline

New? I got my V2 and V2 Neo mid December 2022 and compiled my own binary.

### Expected behavior

I expect the "Change Filament" menu to correctly behave with a DWIN/DACAI screen with a JyersUI.

When you select change filament, the printer should unload the filament, prompt for new filament load, purge, purge (more, if requested), then return the screen to user control. 

It currently stays at the "waitin for user input" screen

### Actual behavior

When you select change filament, the printer should unload the filament, prompt for new filament load, purge, purge (more, if requested), it stays at the "waitin for user input" screen.

### Steps to Reproduce

1. Go to Prepare --> Change Filament
2. Printer unloads filament
3. User pulls out old filament
4. Printer prompts to load filament
5. User loads new filament
6. Click continue.
7. Printer purges filament.
8. Printer asks to purge more (continue), or stop (cancel).
9. Clicking continue to purge more causes the screen to display "waiting for user input" after the second purge is complete. Printer is then stuck. Recovery is through power cycle.

### Version of Marlin Firmware

Marlin release 2.1.2

### Printer model

Ender 3 V2 + V2 Neo

### Electronics

4.2.2 board

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Configuration files for my Ender 3 V2 and Ender 3 V2 Neo are attached.
[Github.zip](https://github.com/MarlinFirmware/Marlin/files/10419733/Github.zip)

RV-from-be, 2023-01-16:
@ngth82
You will find a fix at #24996. Unfortunately the Marlin team does not always take this into account and seems to have dropped the jyersui in favor of the proui. What does @Jyers (Jacob) think?

EvilGremlin, 2023-01-16:
Jacob is notably absent from marlin development for last year or so, MIguel is present and doing stuff. Anyone is free to pick up JyersUI maintenance and development if you think it should be kept as separate entity.

closed 2023-01-16 by thisiskeithb

ngth82, 2023-01-16:
Hi guys - good morning!

Thank for looking and triaging. But I think this issue is separate from #24996. 

I have already implemented the fixes to dwin.cpp, encoder.cpp, and pause.cpp (see attached zip file).  #24996 is a M600 related issue when printing. This bug that I'm filing is for the screen getting stuck when you go to Prepare --> Change Filament using a DWIN/DACAI display and the JayersUI. It just won't return to a usable UI.

I've attached my dwin.cpp, encoder.cpp and pause.cpp files in this response. You'll see the proposed patches for these three files in #24996  is already present. 

Workarounds can just be going to the preheat PLA menu instead, but certainly i'm not the only one that will (eventually) encounter this.

[cpp_files_25242.zip](https://github.com/MarlinFirmware/Marlin/files/10427633/cpp_files_25242.zip)

ngth82, 2023-02-28:
> @ngth82 You will find a fix at #24996. Unfortunately the Marlin team does not always take this into account and seems to have dropped the jyersui in favor of the proui. What does @Jyers (Jacob) think?

@Jyers  - Did you have an opportunity to look at this? The Marlin dev team says this is a duplicate of similar issues, but I don't think it is.