MarlinFirmware/Marlin #25243: [FR] (Part cooling fan start/stop time compensation)

dp250f, 2023-01-15:
### Is your feature request related to a problem? Please describe.

When printing a part where the part cooling fan only turns on when bridging or printing overhangs, the part cooling fan takes too long to start and stop. On shorter overhangs and bridges, the fan ends up not cooling the part it should and cooling something it shouldn't.

### Describe the feature you want

It would be nice to have a feature which can start and stop the part cooling fan x.x seconds before the G-Code calls for it so the fan is more likely to cool the exact part of the print you want it to.

### Additional context

I often see this part cooling delay artifact in the first part of a bridge where it droops more in one corner than the rest of the bridge and you can't fix it with slicer settings.  This becomes an issue when printing materials like CF-PA6 where you only use part cooling for bridges.

Bob-the-Kuhn, 2023-01-15:
I don't see that this can be done reliably in Marlin.  It would require Marlin to look back through it's command queue, calculate build times and then inject the fan control sequence there.  Besides CPU usage concerns, there comes a point where the x.x seconds setting is so large that the injection point has already been executed.

This might be possible to do in the slicer software.

An issue with both is the accuracy of the build time calculation.  My printer tends to take about 2x the time estimated by the slicer.

Nuck-TH, 2023-01-18:
Prusa and Super Slicers have this already.
It is not really possible to do in firmware, since while Marlin has command buffer and lookahead, buffer is relatively short(due to MCU RAM constaraints) and obviously fits certain length of (variable length) commands, not certain time frame of them. It can vary from couple of milliseconds for series of short moves to tens of seconds for long blocking commands like G28(home).

thisiskeithb, 2023-01-18:
Since this is not possible/should be done in the slicer, closing.

closed 2023-01-18 by thisiskeithb