MarlinFirmware/Marlin #25246: Marlin Configuration tool

gsuresh2u, 2023-01-16:
### Is your feature request related to a problem? Please describe.

Hi, this is my first project in Python
done some work configuration tool, its working fine
i think i have completed more than 90% of work
[image]
[image]
[image]
[image]
[image]

[image]

### Are you looking for hardware support?

after modifying the congih files with this tool, i found a problem in VS code
that is, when we enable a feature like LIN_ADVANCE, under that options, the lines are still not showing in bright color
please check the attached image
[image]

### Describe the feature you want

if anyone try to implementt this tool, i am attaching the all source code
all drivers, motherboards, etc.. are taking from txt files
[Marlin Configurator.zip](https://github.com/MarlinFirmware/Marlin/files/10426064/Marlin.Configurator.zip)

### Additional context

Please help me on one thing
In configuration_adv.h file, if we enable #define CONFIG_EXPORT 2 .
it is exporting present configuraion tto file config.ini
if we download any new version is there any option to import that config.ini to new version ?
please let me know
If there is option to import that confog.ini to new version, it will be very usefull to all Marlin users

EvilGremlin, 2023-01-16:
This is what Scott doing with ABM plugin, all late changes in configs formatting etc are so plugin can have universal parser. But i like your layout ideas more, because, well, there is layout present.
> problem in VS code

this is normal, intelisense being dumb

thisiskeithb, 2023-01-16:
As mentioned above, there is already a tool in place: https://github.com/MarlinFirmware/AutoBuildMarlin

closed 2023-01-16 by thisiskeithb

gsuresh2u, 2023-01-17:
But Auto build Marlin is not configuration tool
It just build firmware

Is there any tool available to import Config.ini file settings to new version of Marlin?

thisiskeithb, 2023-01-17:
> But Auto build Marlin is not configuration tool
It just build firmware

it does. Take a look at the other branches: https://github.com/MarlinFirmware/AutoBuildMarlin/branches

gsuresh2u, 2023-01-17:
Sorry for my ignorance
Can you please give more details on how to use that