MarlinFirmware/Marlin #25247: [BUG] SKR v2B onboard SD not working

VanessaE, 2023-01-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Somehow, in the aftermath of some recent update, I've lost access to both the motherboard's onboard µSD card slot on my SKR v2-based printer, and the 12864 display module's SD card slot.

The onboard µSD is supposed to show up on my PC as a common USB mass storage device, and the LCD SD is supposed to be hidden from the PC, only appearing within Marlin's menus as a source to print from.  In fact, that the motherboard's µSD card appeared as mass storage is how I would normally update the firmware -- just compile and copy `firmware.bin` to the card in the printer, as if it's a USB card reader.

Instead, the two cards/slots just don't work, at least not in all of the ways they're supposed to.

If I insert the µSD card (Verbatim 8 GB, class 10) into my PC's card reader, I can drop a suitable `firmware.bin` onto it and THEN move that card to the offending SKR v2, I can then reset or power-cycle the printer, and the SKR's bootloader will happily read the card and flash the new firmware exactly like it should.

Starting with the configs I've attached below, if I uncomment `#define SDCARD_CONNECTION LCD` (thus overriding Marlin's default of `ONBOARD`), and set the build environment to `BIGTREE_SKR_2`, the LCD SD card slot works, and I could print from there if I wanted to.

If I uncomment that, and set the environment to `BIGTREE_SKR_2_USB`, I can access the LCD SD card from the PC as mass storage, and I can also tell Marlin to print from it.  However, that can crash Marlin (had that happen once already, this might be related to #25013).

With some other settings changes, I can also make the SKR v2's thumbdrive port work, and a drive plugged in there shows up on the PC and inside Marlin.

I have tried every combination of settings that I can think of that might be related, but no case can I get the SKR's onboard µSD card to work in Marlin.  I even tried turning the `SD_SPI_SPEED` down to half, then quarter speed, to no avail.  All that seemed to do was make Marlin slower at startup.

Something of possible interest, look at this dmesg output from when I turn on the offending printer:
 
```
[  +0.351948] usb 3-2.1.4: new full-speed USB device number 85 using xhci_hcd
[  +0.148064] usb 3-2.1.4: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 0.00
[  +0.000003] usb 3-2.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 3-2.1.4: Product: MARLIN_STM32F407VGT6_CCM CDC in FS Mode
[  +0.000000] usb 3-2.1.4: Manufacturer: STMicroelectronics
[  +0.000001] usb 3-2.1.4: SerialNumber: 3958366C3439
[  +0.049030] cdc_acm 3-2.1.4:1.0: ttyACM0: USB ACM device
[  +0.011982] usb-storage 3-2.1.4:1.2: USB Mass Storage device detected
[  +0.000143] scsi host6: usb-storage 3-2.1.4:1.2
[  +1.033061] scsi 6:0:0:0: Direct-Access     MARLIN   Product          0.01 PQ: 0 ANSI: 2
[  +0.000150] sd 6:0:0:0: Attached scsi generic sg0 type 0
[  +0.030029] sd 6:0:0:0: [sda] Attached SCSI removable disk
[  +6.048428] sd 6:0:0:0: [sda] 3521118208 512-byte logical blocks: (1.80 TB/1.64 TiB)
```

Umm... no...  it's only an 8 GB card.... in fact no storage device or partition on my system measures 1.8 TB. :smile:   The interesting bit is that that reported volume size is consistent -- it's precisely 3521118208 every time I reset or power-cycle the printer.

On the other hand, both cards/slots work fine in my other printer, which uses an SKR v1.4.  Over there, the LCD SD is hidden from the PC, but shows up in Marlin's menus as something I can print from, and the onboard µSD card shows up on the PC as a normal mass storage device which I can use for firmware updates.  This is a completely ordinary setup.

Since the SKR v2 can flash, and I can get the LCD SD working by "improperly" configuring the firmware as describe above, and since both slots/cards work fine in the other printer, that would seem to rule-out the possibility of bad SD cards or motherboard failure.

After some chat on Discord, one user suggested this is a fault in Marlin's SDIO access, but he couldn't offer any help beyond that.

As a sanity check, I dug out an older firmware from my backups and flashed it to offending motherboard.  THAT firmware works.  The onboard µSD shows up as both mass storage and in Marlin's menus as something to print from, while the LCD SD is inoperative.

Access to the µSD is sluggish and prone to causing screen garbage, but others have reported that already (and I had that happen with the LCD SD with today's firmware, when I had that dual mode enabled during testing).

Oddly, the dmesg output when I power-cycle/reset the printer with this old firmware also shows that oddball 3521118208/1.8 TB thing I mentioned above, no idea at all where that's coming from, but aside from that and the sluggish access, it's completely fine.

### Bug Timeline

It's old, started sometime after 2021-10-18

### Expected behavior

I expected the onboard µSD card to show on my PC as a mass storage device, and for the LCD SD to be available within Marlin (hidden from the PC), just as on the other printer with the SKR v1.4.

### Actual behavior

They...  er... didn't show up.

### Steps to Reproduce

No idea how someone else would be able to reproduce this, if compiling with my configs alone isn't enough.

### Version of Marlin Firmware

bugfix-2.1.x commit d94defc5

### Printer model

custom Hypercube

### Electronics

SKR v2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Configs for current Marlin, with onboard µSD not working:
  [Marlin-configs-2023-01-16.zip](https://github.com/MarlinFirmware/Marlin/files/10427640/Marlin-configs-2023-01-16.zip)

Configs for the old firmware I dug out, where onboard µSD does work:
 [Marlin-configs-2021-10-18.zip](https://github.com/MarlinFirmware/Marlin/files/10428274/Marlin-configs-2021-10-18.zip)

Bob-the-Kuhn, 2023-01-16:
Does you motherboard have the STM32F407VG or STM32F429VG cpu?

VanessaE, 2023-01-16:
It is the F407 variant

(board revision B allegedly :stuck_out_tongue: )

VanessaE, 2023-01-18:
All right, I'm gonna chalk this up to maybe a PlatformIO update somewhere that I didn't notice, because I recompiled and flashed with the very same code and config that wasn't working and prompted me to file this issue, and now the onboard µSD is working.  It's in that dual-access mode as with the old firmware, so PC access to the mass storage device is sluggish, but it seems to work just fine.

Or it could be some intermittent hardware failure that isn't failing now... but that would be par for the course with SKR v2.  Frankly this motherboard is junk, so if that's the problem it wouldn't surprise me.

I'll leave this issue open for now, in case it really IS a bug that simply isn't...re ...bugging out.

thisiskeithb, 2023-02-24:
> I re-flashed with the very same code and config that wasn't working before (that prompted me to file this issue), and now the onboard µSD is working

> I'll leave this issue open for now, in case it really IS a bug that simply isn't... er... bugging out.

No issues here, so closing.

closed 2023-02-24 by thisiskeithb