MarlinFirmware/Marlin #25248: [BUG] Processor Death

aviatorhh, 2023-01-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After applying latest bugfix built, my ATmega2560 on the MKS Gen L V2.0 lost his life.
I do not really think it is caused by the software alone but want to set this as documented. It seems the case fan has quit so overheating may be the cause. Anyhow I did not expect to have an outcome like this. That centipede seems to work hard while printing (and therefore a software bug?).
My fix will be some kind of case fan monitoring and improved cooling.

[image]

### Bug Timeline

kinda new

### Expected behavior

Not to die in flames.

### Actual behavior

The processor has lifted its head after some printing time. It was not the first print after the recent flashing but close to.

### Steps to Reproduce

1. compile 
2. flash
3. run
4. die (after some time printing and not the first after flash)

### Version of Marlin Firmware

I cannot run M115 anymore but I have bugfix-2.1.x 212167b140

### Printer model

Anet A8

### Electronics

MKS GEN L V2.0

### Add-ons

Two extra FETs for HE and Bed; RGBW (via transistors); Case LED (via transistor)

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Slicing done with Fusion 360
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/10428331/Archive.zip)

borland1, 2023-01-18:
If MCU were to run hot, it would have a heat sink attached. The chassis fan is for the stepper motor drivers which typically run hot and have heat sinks   Power supply fan is commonly always-on or self-regulated.

EvilGremlin, 2023-01-18:
It's impossible for any chip to explode because "it work too hard". You had major electrical issue that pumped all your main 12/24V power into MCU.

ellensp, 2023-01-19:
a common failure is hotend wiring...   hotend heater wires shorting due to wearing to the hotend, when one of the legs of thermistor is already shorted to it.  or vice versa...   either way high voltage jumps to the cpu.