MarlinFirmware/Marlin #25251: [BUG] Too much fixes needed since PRO UI MPC menus  #25232 : Need finition/Not ready

studiodyne, 2023-01-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

@thinkyhead 
Impossible to compile because of the big commits PRO UI MPC menus 
Preprocessor/Macro/class / members

You can use my  TronxyV7 , try to compile and see where to fix

Thks

### Version of Marlin Firmware

bf2.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10451324/Marlin.zip)

ellensp, 2023-01-19:
none of the errors so far are related to code.

 1066 |     #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."
 1274 |     #error "SERVO1_PIN must be defined for your SWITCHING_NOZZLE."
 1289 |       #error "SERVO2_PIN must be defined for your SWITCHING_NOZZLE."
 1759 |       #error "SERVO0_PIN must be defined for your servo or BLTOUCH probe."
 1808 |       #error "BLTOUCH requires DEACTIVATE_SERVOS_AFTER_MOVE to be to disabled. Please update your Configuration.h file."
 
All of these are configuration issues

You haven't told marlin where you have all these things plugged in.

ellensp, 2023-01-19:
Your FILAMENT_RUNOUT_SCRIPT is utter garbage

you have #define FILAMENT_RUNOUT_SCRIPT "M7/n G4 S2/n M9/n M125T"

Firstly you use \n for command separators not /n
T on is own is not a valid g-code
spaces are not needed

I suspect this is what you intended.

#define FILAMENT_RUNOUT_SCRIPT "M7\nG4S2\nM9\nM125T%c"

ellensp, 2023-01-19:
Having this disabled PREVENT_COLD_EXTRUSION is showing a bug/

Marlin/src/gcode/lcd/M145.cpp: In static member function 'static void GcodeSuite::M145()':
Marlin/src/gcode/lcd/M145.cpp:50:72: error: 'class Temperature' has no member named 'extrude_min_temp'
   50 |         mat.hotend_temp = constrain(parser.value_int(), thermalManager.extrude_min_temp, thermalManager.hotend_max_target(0));

This code presumes extrude_min_temp exists 
```
    #if HAS_HOTEND
      if (parser.seenval('H'))
        mat.hotend_temp = constrain(parser.value_int(), thermalManager.extrude_min_temp, thermalManager.hotend_max_target(0));
    #endif
```

But extrude_min_temp is only setup when PREVENT_COLD_EXTRUSION is defined
```
    #if ENABLED(PREVENT_COLD_EXTRUSION)
      static bool allow_cold_extrude;
      static celsius_t extrude_min_temp;

```

ellensp, 2023-01-19:
PR https://github.com/MarlinFirmware/Marlin/pull/25232 did intraduce in issue
replaced constants with mpc, but missed one in Marlin/src/lcd/menu/menu_advanced.cpp

causing
```
Marlin/src/lcd/menu/menu_advanced.cpp:380:76: error: 'hotend_info_t' {aka 'struct MPCHeaterInfo'} has no member named 'constants'
  380 |           MPC_t &mpc = thermalManager.temp_hotend[MenuItemBase::itemIndex].constants; \
      |                                                                            ^~~~~~~~~

```
simply rename constants to mpc

ellensp, 2023-01-19:
PR created to fix the issues.

closed 2023-01-19 by ellensp

studiodyne, 2023-01-19:
Thank you very much , good job

Filament runout script explaination
"M7/n G4 S2/n M9/n M125T"

M7  ' connect to simcard board to send SMS with ' pause ' message 
G4 to pause2 seconds before
Output state recover M9 
M125 T ' go to pause ' ,   this is 'MY ' pause , the best than others , try it , look my PR ' not merged yet '  , the best pause ever lol

Simple !

ellensp, 2023-01-19:
/n is still wrong...  it is meant to be \n  ie a newline character

GMagician, 2023-01-19:
> M125 T ' go to pause ' 

Not so sure but T should be reserved to tool change in gcode and can't be used as parameter, more or less like G and M

studiodyne, 2023-01-19:
Make corrections/fixing , and give it to the communauty for intensive testing
We need a serious pause , with available menus , to clean nozzle manually , to change nozzle , to park somewhere we can insert a wire in the nozzle ,   M125 T or any letter , is not a feature , it's simply the ' base ' pause , for a robotic machine that need human intervention anytime during a print

Thanks for the ' \n' :smile ,  pause testing if for tomorow , you just give me one or two hours of searching issues lolllllllllll