MarlinFirmware/Marlin #25260: [FR] (feature summary) UBL mesh edit LCD interface

apulver, 2023-01-21:
### Describe the feature you want

When editing a mesh using "Mesh Edit" from the UBL menu, it would be convenient if, after editing a mesh value, the user were returned to the graphical "grid" where points can be viewed and selected, rather than automatically moved along to the next adjacent point to edit.

borland1, 2023-01-22:
Are you sure about that?  This 3-year old Chris Riley YouTube.com video shows him using UBL edit mesh with LCD menu..  each point selectable individually. 

[Marlin Unified Bed Leveling - How To - Chris's Basement](https://www.youtube.com/watch?v=ONpKxkil16Q&t=376s)

[image]

apulver, 2023-01-24:
> Are you sure about that? This 3-year old Chris Riley YouTube.com video shows him using UBL edit mesh with LCD menu.. each point selectable individually.

Yes, you can select each point individually, but then you are taken to every adjacent point. Say you want to edit a row or column, e.g. if your x-axis does not have enough travel to probe the rightmost column of the bed. It would be convenient to have the option of editing just one single point before the user is returned to the grid.