MarlinFirmware/Marlin #25261: Z_MULTI_ENDSTOPS - Not working

VenoMexx, 2023-01-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to configure my mks nano 3.1 with two independent Z-Axis Motors and two Endstops.
I defined these endstops:

> #define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_ZMAX_PLUG

drivers:

> #define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define Z2_DRIVER_TYPE TMC2209
#define E0_DRIVER_TYPE TMC2209

And enabled Multi Endstops:

> #ifdef Z2_DRIVER_TYPE
// #define INVERT_Z2_VS_Z_DIR // Z2 direction signal is the opposite of Z
#define Z_MULTI_ENDSTOPS // Other Z axes have their own endstops
#if ENABLED(Z_MULTI_ENDSTOPS)
#define Z2_USE_ENDSTOP _ZMAX_   // Z2 endstop board plug. Don't forget to enable USE_*_PLUG.
#define Z2_ENDSTOP_ADJUSTMENT 0 // Z2 offset relative to Y endstop
#endif

M119 (all endstops working)

> x_min: open
y_min: open
z_min: open
z_max: open

z_min endstop can stop z0 driver but z_max cant stop z1 driver.

Can anyone help me with this issue?

### Version of Marlin Firmware

bugfix-2.1.x

### Electronics

MKS robin nano 3.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10471774/Configuration.zip)

ellensp, 2023-01-21:
You say you have a MKS robin nano 3.1 but you set #define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3
not BOARD_MKS_ROBIN_NANO_V3_1 

Make sure the Z+ diag pin jumper is off, as this will interfere with endstops

VenoMexx, 2023-01-21:
> You say you have a MKS robin nano 3.1 but you set #define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3 not BOARD_MKS_ROBIN_NANO_V3_1
> 
> Make sure the Z+ diag pin jumper is off, as this will interfere with endstops

i try but some problem continue

ellensp, 2023-01-21:
You have reformatted the config file, making it impossible to see what you have and have not changed,  Ever line has been changed!

VenoMexx, 2023-01-21:
> You have reformatted the config file, making it impossible to see what you have and have not changed, Every line has been changed!

i updated

ellensp, 2023-01-21:
you have Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

you cannot have a probe on the Z-min, while using Z-min for homing Z1

When you use real marlin 2.1.x bugfix, it tells you this

```
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2816:6: error: #error "Z_MULTI_ENDSTOPS is incompatible with USE_PROBE_FOR_Z_HOMING."
 2816 |     #error "Z_MULTI_ENDSTOPS is incompatible with USE_PROBE_FOR_Z_HOMING."

``` 
Which is not particularly clear what you have to do in this case

Setup a Z_MIN_PROBE_PIN instead.

closed 2023-01-21 by ellensp