MarlinFirmware/Marlin #25263: [BUG] "echo:busy: paused for user" after filament runout trigger

Papadoma, 2023-01-22:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I have a smart filament runout sensor from BTT installed; if the filament runs out or gets jammed, the print halts and waits for the user to check it.
At the same time I have the Makerbase MKS H43 as an interface screen. Ive removed the previous graphical LCD with the knob. As such Im aware of certain issues with such screens, like for example the fact that they don't work that good with the runout functionality.

However, on the previous 2.0.9.3 I could purge and continue the print after the filament change using the web UI. Now, even though I get the same promts, the printer won't continue. It says "paused for user" continuously.

### Bug Timeline

after updating to 2.1.2

### Expected behavior

Be able to continue the print either via the web UI or the Makerbase MKS H43 screen.

### Actual behavior

Wasted a couple of prints as I couldn't continue.

### Version of Marlin Firmware

2.1.2

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/10473460/Configuration_adv.zip)

Papadoma, 2023-01-22:
I had another go at this and I think the reason why it wouldn't allow me to continue is because I had forgotten to enable 
`#define EMERGENCY_PARSER`
I did so let's see.

However, I also noticed that there is a specific mention of the runout sensor functionality on H43's wiki page
[https://github.com/makerbase-mks/MKS-H43/wiki/Filament-runout-configuration](url)
I tried enabling it and now it changes the UI to a different one upon printing, which was unexpected. However, upon starting a print, I immediately get a prompt on the UI that the runout sensor was triggered, which is unexpected and thus a bug. 
As instructed by the wiki, I tried without the `#define FILAMENT_RUNOUT_SENSOR`. Didn't work, the runout is triggered always on start and never afterwards. Then I tried enabling `#define FILAMENT_RUNOUT_SENSOR`, same behaviour. 

Not sure why this screen does not work correctly with marlin.

Papadoma, 2023-04-03:
Issue still occuring, so not stale.
Enabling the emergency parser allowed me to change the filament and continue
However, I cannot make the screen work properly with the BTT smart filament runout sensor. The problem is not resolved

Papadoma, 2023-05-26:
Bad bot!

closed 2023-08-06 by github-actions[bot]

thinkyhead, 2023-10-06:
MKS UI has its own filament runout handling which is not properly integrated with Marlin yet. You can try disabling the standard Marlin runout feature and enable `DGUS_MKS_RUNOUT_SENSOR` instead. See #22595