MarlinFirmware/Marlin #25264: [BUG] Controller restart when printing with MIXING_EXTRUDER enabled.

yoch2015, 2023-01-22:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Running a test print with MIXING_EXTRUDER enabled on a Delta machine with 3 extruders resets the controller with G0 F2700 X-8.863 Y11.603 Z0.3 on line 19 of gcode.
[CFFFP_test_circle.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/10473997/CFFFP_test_circle.gcode.zip)
[config.zip](https://github.com/MarlinFirmware/Marlin/files/10473998/config.zip)

### Bug Timeline

today

### Expected behavior

normal running.

### Actual behavior

controller restart.

### Steps to Reproduce

1.Print from the Pronterface.

### Version of Marlin Firmware

2.1.2

### Printer model

Generic Kossel

### Electronics

BTT SKR v1.3

### Add-ons

external stepper (E2) ; PT100

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-01-22:
You need to provide sufficient files for us to recreate it.
You have added several extruders, so have a modified boards pin.h file
please provide this file also. Also any other marlin files you have edited.

yoch2015, 2023-01-22:
I have attached the pin file.
I also ran M111S247 and tested again, no controller reset occurred.
The event did not occur even after retesting after M111S0.
The difference from the previous step is probably turning auto_fan on and off.
The reproducibility has become poor, so closing is fine.
[pins.zip](https://github.com/MarlinFirmware/Marlin/files/10474101/pins.zip)

ellensp, 2023-01-22:
you do have 

#define E0_AUTO_FAN_PIN P2_03
#define E1_AUTO_FAN_PIN P2_03
#define E2_AUTO_FAN_PIN P2_03

but only have 
#define TEMP_SENSOR_0 20

So E1_AUTO_FAN_PIN and E2_AUTO_FAN_PIN might be causing you issues

yoch2015, 2023-01-22:
Thank you very much.
I agree.
The E-factor has also been modified to 1, so the fan pin is likely to be affected as well.

yoch2015, 2023-01-22:
After changing Configuration_adv.h and testing continuous printing before turning auto_fan off, it seems to have resolved the issue.
Thank you very much.

closed 2023-01-22 by yoch2015