MarlinFirmware/Marlin #25265: [FR] Add standard peripheral serial port defines for all motherboards. Makes config file easier for users.

ellensp, 2023-01-22:
### Is your feature request related to a problem? Please describe.

Users seem to easily get confused as to what serial port numbers to use.

Developers and support people know to look at board manufacturers pin diagram or to look at circuit diagrams, , But they don't always name the UART port used, sometimes it is IO pins, which the user is not expected to be able to translate.

   

### Are you looking for hardware support?

no

### Describe the feature you want

Add the following defines per board.  

CONTROLLER_USB_SERIAL_PORT
CONTROLLER_TFT_SERIAL_PORT
CONTROLLER_LCD_SERIAL_PORT
Any others?

Then you can have #define SERIAL_PORT CONTROLLER_USB_SERIAL_PORT
and #define LCD_SERIAL_PORT CONTROLLER_LCD_SERIAL_PORT

Without the user having to worry about what port they are on
While still allowing for numeric settings for customization, for users that know what they want.

Roxy-3D, 2023-01-22:
This makes a lot of sense...

EvilGremlin, 2023-01-22:
+1, thinking about it for a long time

ellensp, 2023-01-22:
HMI_SERIAL_PORT may be technically correct, but no one, including myself, knows what that means!

Bob-the-Kuhn, 2023-01-22:
Along similar lines ...

How about adding info to configuration.h indicating which LCDs use SERIAL_PORT and which use LCD_SERIAL_PORT.

ellensp, 2023-01-22:
There are no LCD types in marlin config that use SERIAL_PORT, only those that need LCD_SERIAL_PORT

EvilGremlin, 2023-01-22:
It's naming hell. TFT refer to what? DWIN is TFT alright. Unledd they have TN models which i doubt. 
Maybe
BOARD_USB_SERIAL_PORT
BOARD_5PIN_SERIAL_PORT
BOARD_EXP_SERIAL_PORT (for creality 10-pin mostly)
BOARD_WIFI_SERIAL_PORT

ellensp, 2023-01-22:
It doesn't really matter, what ever we come up with thinkyhead will change it, 

So for now I say we use temporary names

OPAL_PORT    for usb
JADE_PORT    tft port    
RUBY_PORT   for wifi
ONXY_PORT  for exp

Then a simple search and replace can be done for the final name that passes thinkyhead scrutiny.

EvilGremlin, 2023-01-22:
For now i vote fot BOARD_#PIN_SERIAL_PORT, numbers can be 3,4,5 and 10. And wrong serial error must link to doc explaining stuff (including referencing datasheets etc)

ellensp, 2023-01-22:
Well if your not going to be serious.

EvilGremlin, 2023-01-22:
?? i'm quite serious about that naming with pin count. It's representative of board connectors, no matter silkscreen labels and devices plugged.

Roxy-3D, 2023-01-23:
This is a good idea and we need to work through the issues.

Roxy-3D, 2023-01-24:
One question I have that could use some discussion is this:     

If a board has multiple serial ports available, should they all be enabled by default?   Or should just one be enabled unless the user specifies multiple are desired?    And if only one is enabled, my thinking is it should be the USB Serial port that has hardware support (assuming that exists).

The-EG, 2023-01-24:
At least one case of not wanting all available serial ports enabled comes to mind: Creality v4.2.x boards have a UART in the EXP port. This is so that the DWIN screen (ie. Ender 3 v2) can be used, but you obviously can't use that UART if you are using something else, like a `CR10_STOCKDISPLAY`. 

I also really like this idea. This was one of the hurdles for me when upgrading to the BTT Octopus board, and trying to get the above DWIN screen working. I had no idea that the UART numbers were actually labeled as RXn/TXn on BTTs pin diagrams until *after* I had stumbled upon the correct UART number and then made the connection. I got it working by just trying numbers until it worked.

I also think the most helpful and easy to understand approach, for the ***average user***, would be to name them as they are on the board/documentation. I believe that's also what the original intent with the changes here was.

With the Octopus you'd end up with something like:
```c++
#define CONTROLLER_TFT_SERIAL_PORT 1
#define CONTROLLER_CAN_SERIAL_PORT 1
#define CONTROLLER_RPI_SERIAL_PORT 2
#define CONTROLLER_WIFI_SERIAL_PORT 3
```

You can argue all you want about what RPI and WIFI should be called, but in reality a user looking at either of those is going to see things like this:
[image]

Roxy-3D, 2023-01-25:
The BTT Octopus board is a good example to use.  I have a BTT octopus board and its serial ports have to be messed with to get it to work reliably with PronterFace.    We should probably have a couple of other 'problem' boards that we check our thinking against.   Maybe one of the Creality boards too?

closed 2023-02-05 by Roxy-3D