MarlinFirmware/Marlin #25266: [FR] Add support for TriGorilla Gen_V3.0.6 (Anycubic Kobra GO/NEO)

cringegnere, 2023-01-22:
### Is your feature request related to a problem? Please describe.

Need to add support for Trigorilla V3.0.6

### Are you looking for hardware support?

Printer: Anycubic Kobra Neo / GO (I own a Neo, but GO has the same motherboard)
Board: Trigorilla V3.0.6

### Describe the feature you want

Add support for Trigorilla V3.0.6 for Anycubic Kobra Neo / GO

### Additional context

The Anycubic italian website has the motherboard in the shop (with pictures)
https://anycubic.it/products/scheda-madre-per-anycubic-kobra-go?_pos=7&_sid=e5a7747cc&_ss=r
[image]

Should have a Hc32f460kct6 processor (ref. https://github.com/MarlinFirmware/Configurations/issues/649#issuecomment-1376998875)

EvilGremlin, 2023-01-22:
Not in any forseeable future, for now you're stuck with building in Keil.

cringegnere, 2023-01-22:
May I ask why? I'm just curious on how high the effort would be

EvilGremlin, 2023-01-22:
Gotta wrote platformio/arduino core for these HDSC chips first.

cringegnere, 2023-01-22:
Based on your experience, how much would take to update Marlin from the Anycubic base repo with Keil? I would like to try to do it but i'm no firmware expert

EvilGremlin, 2023-01-22:
At first glance you'll have to port all of their meaningful changes (display and HAL). First rename HAL and tweak stuff to not conflict with vanilla of same version, then merge. No idea if Keil can do proper source filtering, probably gotta setup gitignore first.

EvilGremlin, 2023-01-22:
If you can maintain up-to-date fork and prebuilts for these machines we will be very very grateful.

1coderookie, 2023-02-03:
Hi, 
I have no idea if this might be helpful somehow as I don't know enough about this whole firmware stuff, but a user on reddit mentioned that Klipper supports this chip now and he got it running on this particular TriGorilla V_3.0.6 board, so maybe that could be helpful somehow for Marlin also as you guys might get some useful informations out of the Klipper integration..?
Here's the reddit post about it: https://www.reddit.com/r/anycubic/comments/10cwm16/install_klipper_on_kobra_go_or_neo/

EvilGremlin, 2023-02-03:
No, klipper is very easy to port and completely irrelevant. One precompiled binary fits all configuration, even chips with different pin count.

1coderookie, 2023-02-03:
Oh ok, sorry, I didn't know that :(

cringegnere, 2023-02-03:
I switchet to Klipper, which supports the motherboard and does not have problems with the driver. Now i can print slowly even with a 0.2mm nozzle withouth get the extruder to stop middle print.
I also enabled pressure advance (aka linear advance for Marlin) and it works great.

Unfortunately Marlin is not easy as Klipper to port, so i will stick with Klipper in the meantime.

NuclearPhoenixx, 2023-11-06:
> Unfortunately Marlin is not easy as Klipper to port, so i will stick with Klipper in the meantime.

From my own experience, compiling the source code provided by Anycubic with Keil is pretty easy, however, even updating to a newer version of Marlin is complete pain. The code they provide is honestly hot garbage with so many trippy modifications to the Marlin core like hard-coded flags that would normally be handled by `Configuration.h` anyways. Endless typos in functions and so on... It's all just pieced together and very confusing.

It surprises me that it works this well when on top of that it gets compiled in some long outdated software with ancient packages. It's a lot easier to just switch to Klipper at this point and I'm certain there won't be another Marlin update for these devices as they clearly didn't even have the time to write the current release. They're very cheap after all.

It would be a miracle to have platformio support for these chips, so I completely understand the state of this issue.

vfiuchcikicshuusrch, 2024-05-27:
there is not hc32f460kcta, but hc32f460keta.
official firmware more than 256kb, that hc32f460kcta have. need more than 256kb flash memory, like of hc32f460keta

maybe anycubic use remarked chips...