MarlinFirmware/Marlin #25269: Thermistor 66 Requires MCLTEA >5

dylanmaloney, 2023-01-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have configured and configured and reconfigured Config_adv.h over and over and it will not resolve this error message when trying to build the firmware.  #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5." I have set it to 5, 10, 4, and so on and it will not change from the exact same error message.

Here is the entire Terminal:

In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2368:6: error: #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
 2368 |     #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2368:6: error: #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
 2368 |     #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2368:6: error: #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
 2368 |     #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_bl24cxx.cpp:32:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2368:6: error: #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
 2368 |     #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2368:6: error: #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
 2368 |     #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_flash.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2368:6: error: #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
 2368 |     #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_if_iic.cpp:32:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2368:6: error: #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
 2368 |     #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_sdcard.cpp:31:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2368:6: error: #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
 2368 |     #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_wired.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2368:6: error: #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
 2368 |     #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\sdio.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2368:6: error: #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
 2368 |     #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\tft\../../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\tft\gt911.cpp:26:
Marlin\src\HAL\STM32\tft\../../../inc/SanityCheck.h:2368:6: error: #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
 2368 |     #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_sram.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2368:6: error: #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
 2368 |     #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
      |      ^~~~~
[... 104 lines cut ...]
E3 v2

### Electronics

Dyze hi-temp thermistor

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin Config Github.zip](https://github.com/MarlinFirmware/Marlin/files/10476616/Marlin.Config.Github.zip)

ellensp, 2023-01-23:
User cannot follow/understand simple error messages.

Enable the following in Configuration_adv.h
#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 5
#define MILLISECONDS_PREHEAT_TIME 30000

This leads to

Marlin/src/lcd/e3v2/proui/dwin.cpp:2683:20: error: 'class Temperature' has no member named 'updatePID'
 2683 |     thermalManager.updatePID();

This is because you enabled MPCTEMP which is not supported in 2.1.2

Currently you require the latest bugfix to use MPCTEMP with proui

closed 2023-01-23 by ellensp

ellensp, 2023-01-23:
Updated config for current bugfix 2.1.x
[bugfix 2.1.x configs.zip](https://github.com/MarlinFirmware/Marlin/files/10476841/bugfix.2.1.x.configs.zip)