MarlinFirmware/Marlin #25270: [BUG] (compiling error with PREHEAT_BEFORE_PROBING)

2probro, 2023-01-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I want to preheat my bed and hotend before probing with my CRtouch. I uncommented `#define PREHEAT_BEFORE_PROBING` and commented "#define PREHEAT_BEFORE_LEVELING" in Configuration.h . If I then compile with the AutoBuildMarlin-plugin in VSC, I get the following error: 

```
Marlin\src\lcd/e3v2/proui/dwin.h:123:21: error: 'LEVELING_BED_TEMP' was not declared in this scope; did you mean 'PROBING_BED_TEMP'?
  123 |   int16_t BedLevT = LEVELING_BED_TEMP;
      |                     ^~~~~~~~~~~~~~~~~
      |                     PROBING_BED_TEMP
```

I think this is a bug in the code. The recommendation to change to "PROBING_BED_TEMP" is correct. Someone should implement a way for the code to use "LEVELING_BED_TEMP" when "#define PREHEAT_BEFORE_LEVELING" and "PROBING_BED_TEMP" when "#define PREHEAT_BEFORE_PROBING"

You can't enable both at the same time. This is thus a bug.

### Bug Timeline

I think an old issue, not sure though

### Expected behavior

It should have been compiled, but because of a bug my code couldn't compile.

### Actual behavior

I got an error saying I should use the variable "PROBING_BED_TEMP" instead of "LEVELING_BED_TEMP"

### Steps to Reproduce

1. Enable PREHEAT_BEFORE_PROBING and disable PREHEAT_BEFORE_LEVELING
2. Compile it
3. Get an error

### Version of Marlin Firmware

2.1.x bugfixes

### Printer model

Creality Ender 3 v2

### Electronics

Stock

### Add-ons

CR-touch

### Bed Leveling

ABL Linear grid

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/10479460/Configuration.files.zip)

ellensp, 2023-01-23:
What you describe is quite impossible in current code.

"Enable PREHEAT_BEFORE_PROBING and disable PREHEAT_BEFORE_LEVELING"

 #if ENABLED(PREHEAT_BEFORE_LEVELING)
    int16_t BedLevT = LEVELING_BED_TEMP;
  #endif

so the line that error is never run when PREHEAT_BEFORE_LEVELING is disabled

Bob-the-Kuhn, 2023-01-23:
The author's "steps to reproduce" is not consistent with text earlier in the post.  The text says "You can't enable both at the same time." but the "steps to reproduce" says to "Enable PREHEAT_BEFORE_PROBING and disable PREHEAT_BEFORE_LEVELING".

With Bugfix-2.1.x I get no errors when I "Enable PREHEAT_BEFORE_PROBING and disable PREHEAT_BEFORE_LEVELING".

With Bugfix-2.1.x I get the following error when both are enabled:
`Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2092:4: error: #error "Disable PREHEAT_BEFORE_LEVELING when using PREHEAT_BEFORE_PROBING."`

This sanity check was added in PR #20383 in late 2020.  The author must be using a really old version of Marlin.  

---

I stumbled into a different error while trying to compile the author's config with Bugfix-2.1.x. 
`Marlin\src\lcd\e3v2\proui\dwin.cpp:3328:51: error: 'onDrawZOffset' was not declared in this scope; did you mean 'onDrawHomeOffset'?`

I'll open a new issue for it.

You'll get this error if:
1. There is a probe.
2. one of the DWIN_ display options are selected.
3. BABYSTEPPING is enabled
4. BABYSTEP_ZPROBE_OFFSET is NOT enabled.

Looks like the **#if** logic has problems.

onDrawZOffset is only used when there is a probe and BABYSTEPPING is enabled.
`#if ALL(HAS_ZOFFSET_ITEM, HAS_BED_PROBE, BABYSTEPPING)`

onDrawZOffset is only defined when BABYSTEP_ZPROBE_OFFSET is enable or when there is no probe and BABYSTEPPING is enabled.
```
#if HAS_ZOFFSET_ITEM
  #if EITHER(BABYSTEP_ZPROBE_OFFSET, JUST_BABYSTEP)
```

2probro, 2023-01-23:
I tried again, I am in marlin 2.1.x bugfixes (latest version, downloaded the folder today) so it's not outdated. I turn on PREHEAT_BEFORE_PROBING and PREHEAT_BEFORE_LEVELING is turned off. 

I try to compile, get an error:

```
In file included from Marlin\src\feature\../module/probe.h:33,
                 from Marlin\src\feature\bltouch.cpp:39:
Marlin\src\feature\../module/../lcd/e3v2/proui/dwin.h:123:21: error: 'LEVELING_BED_TEMP' was not declared in this scope
  123 |   int16_t BedLevT = LEVELING_BED_TEMP;
      |                     ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\feature\../lcd/marlinui.h:49,
                 from Marlin\src\feature\cancel_object.cpp:29:
Marlin\src\feature\../lcd/e3v2/proui/dwin.h:123:21: error: 'LEVELING_BED_TEMP' was not declared in this scope
  123 |   int16_t BedLevT = LEVELING_BED_TEMP;
      |                     ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd/marlinui.h:49,
                 from Marlin\src\MarlinCore.cpp:57:
Marlin\src\lcd/e3v2/proui/dwin.h:123:21: error: 'LEVELING_BED_TEMP' was not declared in this scope
  123 |   int16_t BedLevT = LEVELING_BED_TEMP;
      |                     ^~~~~~~~~~~~~~~~~
```

This is a different error from last time, but still the same variable "LEVELING_BED_TEMP" has an error. If I go to dwin.h and delete this line of code, I can successfully compile the firmware. It seems like that line shouldn't be there. I'm talking about the line of code `int16_t BedLevT = LEVELING_BED_TEMP;` in Marlin\src\lcd\e3v2\proui\dwin.h

2probro, 2023-01-23:
I'm sorry, I made a mistake, I was using a marlin-2.1.x-folder provided by https://github.com/mriscoc/Ender3V2S1

I thought I was using the official marlin folder.