MarlinFirmware/Marlin #25271: [BUG] DWIN_LCD_PROUI compile errors & orphan code

Bob-the-Kuhn, 2023-01-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

There are two sets of configuration.h settings that result in DWIN_LCD_PROUI compile time errors:
- TEMP_SENSOR_BED being set to 0
- having a probe, enabling BABYSTEPPING, and NOT enabling BABYSTEP_ZPROBE_OFFSET

The following line will never be executed because one of the preceeding #elif statements will always be executed:
`MENU_ITEM(ICON_SetHome, MSG_SET_HOME_OFFSETS, onDrawHomeOffset, SetHome);`

### Bug Timeline

Unkown

### Expected behavior

No compile errors

### Actual behavior

**TEMP_SENSOR_BED being set to 0**

This gives the following errors:
```
Marlin\src\lcd\e3v2\proui\dwin.cpp:1251:138: error: 'class Temperature' has no member named 'wholeDegBed'; did you mean 'wholeDegHotend'?
Marlin\src\lcd\e3v2\proui\dwin.cpp:1517:22: error: 'BED_MAXTEMP' was not declared in this scope; did you mean 'STR_T_MAXTEMP'?
Marlin\src\lcd\e3v2\proui\dwin.cpp:1518:30: error: 'struct HMI_data_t' has no member named 'BedPidT'; did you mean 'HotendPidT'?
```

**having a probe, enabling BABYSTEPPING, and NOT enabling BABYSTEP_ZPROBE_OFFSET**

This gives the following error:
`Marlin\src\lcd\e3v2\proui\dwin.cpp:3328:51: error: 'onDrawZOffset' was not declared in this scope; did you mean 'onDrawHomeOffset'?`

### Steps to Reproduce

1. Download Bugfix-2.1.x
2. Enable a probe
3. Enable BABYSTEPPING
4. Leave TEMP_SENSOR_BED as 0
5. Enable DWIN_LCD_PROUI
6. Change MOTHERBOARD to a board that supports the DWIN display
7. Compile

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

N/A

### Electronics

N/A

### Add-ons

N/A

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/10484182/Configs.zip)

Bob-the-Kuhn, 2023-01-23:
**TEMP_SENSOR_BED being set to 0**

The solution for this may be to add a sanity check.

---

**The following line will never be executed because one of the preceeding #elif statements will always be executed:**
`MENU_ITEM(ICON_SetHome, MSG_SET_HOME_OFFSETS, onDrawHomeOffset, SetHome);`

I don't know enough about the menu item to have a good feel as to when it should be enabled.  Prior to PR #25176 it would be executed if BABYSTEP_ZPROBE_OFFSET and BABYSTEPPING are both enabled and there is not a probe.

Right now the **onDrawHomeOffset** routine used by it is only defined when BABYSTEPPING is enabled, BABYSTEP_ZPROBE_OFFSET isn't enabled and a probe is enabled.

MSG_SET_HOME_OFFSETS is defined as "Set Home Offsets"  Looks like there is another other menu item that uses this string.

---

**having a probe, enabling BABYSTEPPING, and NOT enabling BABYSTEP_ZPROBE_OFFSET**

The routine **onDrawZOffset** is only defined when either of the following is true:
- BABYSTEPPING is enabled and a probe is NOT enabled
- BABYSTEPPING is enabled and BABYSTEP_ZPROBE_OFFSET is enabled

The routine **onDrawZOffset** is only used in this line:
`EDIT_ITEM(ICON_Zoffset, MSG_ZPROBE_ZOFFSET, onDrawZOffset, SetZOffset, &BABY_Z_VAR);`

This line is only compiled when BABYSTEPPING is enabled and a probe is enabled.  Prior to PR #25176 it would be executed if either of the following are true:
- BABYSTEPPING is enabled and a probe is NOT enabled
- BABYSTEPPING is enabled and BABYSTEP_ZPROBE_OFFSET is enabled     
- 
MSG_ZPROBE_ZOFFSET is defined as "Probe Z Offset".  Looks like there are two other menu items that use this string.

thisiskeithb, 2023-01-28:
Closing since you've opened a PR:
- https://github.com/MarlinFirmware/Marlin/pull/25304

closed 2023-01-28 by thisiskeithb