MarlinFirmware/Marlin #25274: [BUG] RESTORE_LEVELING_AFTER_G28 not work

stef-ladefense, 2023-01-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After G28, G29, M500 and power off
power on printer, G28, M420V
Bilinear Leveling Grid:
0      1      2      3      4
0 -0.326 -0.278 -0.244 -0.227 -0.235
1 -0.281 -0.215 -0.216 -0.223 -0.252
2 -0.258 -0.229 -0.214 -0.237 -0.266
3 -0.247 -0.235 -0.232 -0.253 -0.254
4 -0.269 -0.208 -0.232 -0.237 -0.283
**echo:Bed Leveling OFF**
echo:Fade Height 10.00

I do not understand
Stephane

### Bug Timeline

one week after test last bugfix-2.1.x

### Expected behavior

bed leveling ON after G28

### Actual behavior

bed leveling OFF after G28
[MarlinAI3MS.zip](https://github.com/MarlinFirmware/Marlin/files/10486431/MarlinAI3MS.zip)

### Version of Marlin Firmware

Marlin bugfix-2.1.x 

### Printer model

Anycubic I3 Mega S

### Electronics

stock

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Bob-the-Kuhn, 2023-01-24:
Are you sure you supplied the correct config files?  The files say you're doing AUTO_BED_LEVELING_BILINEAR with a 4x4 grid but the M420 V command is showing a 5x5.

You have ENABLE_LEVELING_AFTER_G28 enabled so the G28, M420 V sequence should show leveling enabled.

Bob-the-Kuhn, 2023-01-24:
ENABLE_LEVELING_AFTER_G28 is disabled.  

You'll have to issue a M420 S1 to turn on levelling.  After that G28 should keep it on because RESTORE_LEVELING_AFTER_G28 has been enabled.

stef-ladefense, 2023-01-24:
Test this tomorowMerci Envoyé depuis mon appareil Galaxy
-------- Message d'origine --------De : Bob Kuhn ***@***.***> Date : 24/01/2023  15:59  (GMT+01:00) À : MarlinFirmware/Marlin ***@***.***> Cc : Stéphane ***@***.***>, Author ***@***.***> Objet : Re: [MarlinFirmware/Marlin] [BUG] RESTORE_LEVELING_AFTER_G28 not work (Issue #25274) 
ENABLE_LEVELING_AFTER_G28 is disabled.
You'll have to issue a M420 S1 to turn on levelling.  After that G28 should keep it on because RESTORE_LEVELING_AFTER_G28 has been enabled.

stef-ladefense, 2023-01-24:
i read doc of G28/G29, It is not clear and, moreover, there is only one option, during the restore_leveling_after_g28 era.
How currently I find this operation, I do not want to easily edit Gcode to add M420 S1
I want to find the old operation where restore_leveling_after_g28 was subtitled to M420S1

moreover it is necessary to add these two options in the paragraphs G28 / G29 with examples because I did not understand in which case to use one or the other, moreover the very little information gleaned on the internet, are contradictory, additional proof that I am not the only one to understand that by reading the .h configuration file.

💋 
stef

Bob-the-Kuhn, 2023-01-25:
If you change your configuration.h file so that ENABLE_LEVELING_AFTER_G28 is enabled, then leveling will automatically be enabled after G28 if there is a valid leveling matrix.

closed 2023-01-25 by thisiskeithb

stef-ladefense, 2023-01-25:
Thanks everyone, now it works. but please add an explanation in Marlin's web gcode definitions pages

xsuper9988, 2023-03-10:
Are these similar issus?
***
Bilinear Mode(3D print size 220*200*200)
Branch bugfix-2.1.x 20230310
https://www.facebook.com/100000143209637/videos/876767006723207/
***
UBL Mode(3D print size 1048*460*460)
Brance bugfix-2.1.x(20230205)
https://www.facebook.com/groups/3Dtechtalk/permalink/6172950786083163/?mibextid=Nif5oz
***
UBL Mode(3D print size 1048*460*460)
Marlin 2.1.1
https://www.facebook.com/groups/3Dtechtalk/permalink/6035459619832281/?mibextid=Nif5oz
***