MarlinFirmware/Marlin #25277: [FR] (feature summary)

Tinker012, 2023-01-24:
### Is your feature request related to a problem? Please describe.

This request is not related to a problem but rather for hardware support

### Are you looking for hardware support?

I am looking for Hardware support for the newly released Creality E3-Free-runs-silent-motherboard
identified by Creality as  CREALITY_CR4NTxxC10  

This board is a ARM Cortex-M4/STM32F401RET6 with onboard TMC2209's in UART mode.

The files released by Creality for this board are located at https://github.com/CrealityOfficial/E3-Free-runs-Silent-Motherboard/tree/main/example/Marlin

### Describe the feature you want

I would like to see this board added to MarlinFirmware going foreward

ellensp, 2023-01-24:
Looks like this, if anyone is interested

[image]

Bob-the-Kuhn, 2023-04-05:
I've added PR #25636 to implement this request.

I don't have a board to test with.  Would appreciate someone to download the code from the PR and see if it really works.  You'll have to use bugfix-2.1.x config files to test.

thisiskeithb, 2023-04-05:
> I've added PR https://github.com/MarlinFirmware/Marlin/pull/25636 to implement this request.

Closing since a PR has been opened.

closed 2023-04-05 by thisiskeithb