MarlinFirmware/Marlin #25283: [FR] question of staple areas

stef-ladefense, 2023-01-25:
### Is your feature request related to a problem? Please describe.

tray fixed by 4 clamps in the corners

### Describe the feature you want

is there an option for the head to avoid these areas without reducing bands counts all around

merci

Bob-the-Kuhn, 2023-01-25:
You can set the bed size and offsets so that a print won't hit the clips.  This involves setting:
X_BED_SIZE
Y_BED_SIZE
X_MIN_POS 
Y_MIN_POS 
Z_MIN_POS 
X_MAX_POS 
Y_MAX_POS 

You may also want to set a minimun Z height when homing X and Y so that the head doesn't hit the clips.  Set Z_HOMING_HEIGHT for this.

stef-ladefense, 2023-01-25:
ok thank you, that amounts to defining a smaller square with the inaccessible turn suddenly?
Too bad we can't define 4 small areas where the clips are located so that the head doesn't go inside.

plampix, 2023-01-25:
That would require changes to your slicer as well, as it probably assumes a rectangular (or circular) bed.

closed 2023-01-25 by thisiskeithb