MarlinFirmware/Marlin #25289: mesh leveling probing fail on btt skrm3v3+btt tftr35 v3

jarvis710, 2023-01-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

No matter what I do, it always fails at some point; although change the mesh 5x5, 10x10, 15x15. It also doesn't work if I change the dimensions of the bed; I have modified all the parameters and it always fails at some point, both on the board and on the screen. I know that in my configuration I have "unnecessary" things and taking up space, but even so, in the simplest mode the same thing happens.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/10506344/config.zip)

### Bug Timeline

i don't if it's an issue or a bug

### Version of Marlin Firmware

marlin-bugfix-2.1.x

### Printer model

Voxelab aquila x2

### Electronics

btt skr mini3 v3 + btt tft35 v3 + crtouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Bob-the-Kuhn, 2023-01-26:
I expect you're getting a probing failed error.  That's usually caused by **Z_PROBE_LOW_POINT** beeing set too high.  Try lowering it by 2 mm and see what happens.

It also might be your BLTouch.  I stopped using them a couple of years ago because they just weren't reliable enough.

jarvis710, 2023-01-26:
Until now it worked perfectly for me, changing the motherboard does not take 15 days. The problem appears to appear in a certain percentage of mesh completion. (in 5x5 at point 21, in 10x10 at point 72 and in 15x15 at point 196). I had Z_PROBE_LOW_POINT at -3 and there was the problem and at -2 the same.
NOZZLE_TO_PROBE_OFFSET { -44, -7, -2.23 }
PROBING_MARGIN 10
Z_PROBE_LOW_POINT -2
Z_PROBE_OFFSET_RANGE_MIN -10
#define Z_PROBE_OFFSET_RANGE_MAX 10

ellensp, 2023-01-27:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.1.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

jarvis710, 2023-01-27:
These were the results of what you suggest to me; but as you can see g29 does nothing. Every time I've tried the same, same result.
[Connecting....txt](https://github.com/MarlinFirmware/Marlin/files/10515153/Connecting.txt)

jarvis710, 2023-01-27:
from pronterface or cura that was the result just like from the screen console, it's as if it didn't accept the g29 command externally.
if i run the g29 command from the screen interface, pronterface returns this.
//action:notification Probing Point 194/225
//action:notification Probing Point 195/225
//action:notification Probing Point 196/225
//action:notification PIKA-SKRME3V3-BLTUBL-MPC Ready.
//action:notification Doing G29
//action:notification PIKA-SKRME3V3-BLTUBL-MPC Ready.
//action:notification Doing G29
//action:notification PIKA-SKRME3V3-BLTUBL-MPC Ready.
//action:notification Doing G29
//action:notification PIKA-SKRME3V3-BLTUBL-MPC Ready.
//action:notification PIKA-SKRME3V3-BLTUBL-MPC Ready.
//action:notification PIKA-SKRME3V3-BLTUBL-MPC Ready.
//action:notification PIKA-SKRME3V3-BLTUBL-MPC Ready.
//action:notification PIKA-SKRME3V3-BLTUBL-MPC Ready

but on the printer screen, before, a probing failure sign jumped at least 2 times, placing the nozzle on the bed and then continuing, until reaching the same point that I attached before, which would be the end; but on the screen it notifies that:
  bed leveling complete. missing probe points have been smart filled

ellensp, 2023-01-27:
@jarvis710 you have lots of 'crap' in your configs

you have  #define SERIAL_PORT_3 1
but serial port 1 is on the cr10 display port, and you have a cr10 stock display enabled.

you have DISTINCT_E_FACTORS
but only one extruder.

you have UBL_HILBERT_CURVE
when have issue, disable this, it not what could be called essential

You have  #define CONTROLLER_FAN2_PIN FAN0_PIN         // Set a custom pin for second controller fan
FAN0_PIN = FAN_PIN which cannot be used as this is always the part cooling fan

You have #define INPUT_SHAPING_X #define INPUT_SHAPING_Y
Probably a good idea whne you having issues, disable this for now.

You have  #define SDCARD_CONNECTION 
This is just invalid, it needs an parameter

You have #define MEATPACK_ON_SERIAL_PORT_2
But serial port 2 is the tft! No tft speaks meatpack to my knowledge.

Try these cleaned up configs

jarvis710, 2023-02-01:
Sorry for not saying anything, but thank you very much. I have been busy these days and I have not been able to thank you. Although I think it is more of a bad calculation of the mesh since if I print it from the menu on the screen; this is not complete. There is a displacement of about 5cm according to the last time I was able to look at the printer.

closed 2023-02-06 by jarvis710