MarlinFirmware/Marlin #25294: [FR] Ability to wait for heater to saturate

gudvinr, 2023-01-27:
### Is your feature request related to a problem? Please describe.

Right now marlin has `M109` and `M190` to wait for _temperature_.  
Tricky part here is that temperature only measured in one spot.  
While it is probably does not matter for heatblock, which is made of single material (usually) and has relatively small heat capacity so once sensor reports value, that likely means you have somewhat even heat distribution.

That is not exactly true for heatbed and will be noticeable for bigger beds, especially with poor thermal insulation that creates massive heat leakage. So it could take quite a while to heat whole bed.

Often sensor located at the very bottom, close to source of the heat. But we care about top of the bed being hot.  
That makes this more noticeable on thick beds with multiple layers creating additional barrier for heat transfer (e.g. if you have metal-glass or metal-magnet-spring steel-pei, etc).

Winter (and colder times of the year in general) also make situation worse.

In the end, actual heating could take several minutes, dozens of minutes, even, until working surface reaches even coverage.

### Describe the feature you want

I would suggest to add a way to determine such "saturation", mainly for bed but it likely could be used for any heater really.

I heard of this idea from another person who uses RRF. They said that they track PWM duty cycles for bed heater until it reaches certain limit.  
I tried to do some research myself but unfortunately I haven't found publicly available information about implementation in klipper or rrf.

It could be added as additional gcode or as optional parameter for `M190`.

thisiskeithb, 2023-01-27:
Marlin only knows the temperature(s) reported by your thermistors, so this should be handled in your slicer with a [`G4` / Dwell](https://marlinfw.org/docs/gcode/G004.html) command before the print starts.

Example:

```gcode
G4 S300 ; Dwell for 5 minutes
```

gudvinr, 2023-01-27:
I know that you can dwell, but that just means you need to eyeball time.  
That leads to several issues:
- Without thermal camera or poking with good thermal probe at different points you can't be sure if that time is enough
- I think time will change depending on ambient temp and maybe other conditions

What I meant is to check `thermalManager.getHeaterPower` after target temperature is reached and wait for power to stabilize or go below certain point.

This needs an investigation but I don't know how to get that data from marlin.

Nuck-TH, 2023-01-28:
thermalManager.getHeaterPower just returns current PWM value of heater(which is set, surprise, based on temperature measurement of "close to heater" sensor by PID or MPC), not any "measured power". This value will stabilize as soon as temperature near sensor is stable, which gets you where you started.

gudvinr, 2023-01-28:
Nowhere in my FR I mentioned that heated power is anything but PWM duty cycle.  
No need to think that you are smarter than everybody.

EvilGremlin, 2023-01-28:
In the end it comes to eyeball anyway... Not everything should be fixed/crutched in software. If your bed takes so much more time to heat up in cold environment - get better heater or use enclosure. Cheap beds are in general underpowered (to save on PSU) and there were particularly horribble machines where it might even reach equilibrium with very uneven heat (like 20 delta) at 100% duty.

gudvinr, 2023-01-28:
You also missing a point. It's not about bed taking a lot of time to _heat up_ nor about inability to add G4 in gcode.  
Not to mention that I already have good AC heater and it was not an issue in the first place so your assumption is simply wrong.

Like I said in the very beginning, bed being massive naturally takes significant amount of time time to reach equilibrium. But print starts before that, which means bed will continue to expand while first layer being printed. This might also cause similar issues when bed mesh for ABL is being created resulting in incorrect mesh.

Basically only way to do it _now_ in marlin is to look at your temperature plot and randomly decide how much you want to dwell.

> which is set, surprise, based on temperature measurement of "close to heater" sensor by PID or MPC

It doesn't matter how close to heater sensor is when you monitor deviation and not value in the moment.    
When power turned off, heat will spread around. Colder rest of the material is, quicker heat spreads, thus temperature _everywhere_ changes quicker too.  
When temperature stabilizes, heat won't spread as much as before, thus temperature around sensor (wherever it is) also won't drop as much.

[image]

It is a plot of bed temperature and heater power. Each rectangle is about 1m here.  
Sure, power output defined by temperature, but PID tries to minimize temperature fluctuation, which makes them very small very quick because, well, that is what people want.

It is not the case for power. When power fluctuations are about 10%, maybe 20%, you can barely see temperature changing. That is the only reason why I mentioned duty cycle here. It's not about its connection to real output power or whatever. It is just more reliable marker.

And when you have that kind of a marker, it is possible to eliminate eyeballing. And if you can do something automatically rather than manually then why not do it that way?

EvilGremlin, 2023-01-28:
Going by thermal expansion, i never noticed any issues personally. In fact, i start printing even before temp stabilize with 
```
  #define TEMP_BED_RESIDENCY_TIME      3  
  #define TEMP_BED_WINDOW             10 
  #define TEMP_BED_HYSTERESIS         10 
```
and have absolutely no isues with that. Yeah, on consequtive prints without cooldown bed happen to be a smidge higher, but i can just babystep it a bit because i'm here anyway.

mAd-DaWg, 2023-01-31:
I really dont understand the kickback here. Its a really useful quality of life feature.

If you start printing before your temps stabilize, then it sounds very likely that you rely on using extra additives for bed adhesion, or never print on anything much above a 30cm square. For beds significantly larger than this you can end up with certain beds multiple mm's to cm's higher across temperature deltas. Printing before temperature saturation simply isnt an option there, let alone trying to do bed leveling. To say "Just wait 5 minutes" is unreasonable. those without thermal cameras wont have the ability to accurately tell if the bed is ready that day, and the time needed will change drastically day to day

The suggested approach of tracking PWM duty cycles deltas over time is pretty smart, and quite frankly would be a godsend to those with large beds

gudvinr, 2023-02-01:
> 30cm square
> mm's to cm's higher

You mean printers with like 1 meter long axes? Or it should've been μm and mm instead?
Multiple mm expansion, let alone cm, is kinda way too much for any conventional printer that people usually have.

For most people it would be in the range of ~0.1mm or less, so around layer height.  
For printing PLA it shouldn't really be an issue I think, but for more picky filaments expansion might ruin prints.

mAd-DaWg, 2023-02-01:
Yes, printers Ivan Miranda sized and up. It also applies to smaller printers with higher thermal masses though.

Large prints will have the base layers cool by the time thermal saturation is reached, the height differences in the bed will level out, causing whatever is managed to be laid down, to push up off the bed in higher places due to stress in the previously lower parts of the bed, causing the part to either warp, or push itself free from the bed.

Even on bigger prints with sub-1-meter printers using glass beds, you can notice instead of one flat sheen surface, there are multiple surfaces were the previous surfaces pushed off slightly from the bed. When you see these, it is sheer luck that the outer(and lower) parts of the print managed to hold onto the bed. In extreme cases with more serious filaments than pla, the print could warp the bed instead of being released from it, or depending on the thickness, strength asnd type of a glass bed, to shatter, especially if non-borosilicate glass.

mAd-DaWg, 2023-02-01:
A similar situation can happen when the bed is cooling after a print, where the print thermally insulates the bed more than exposed sections of the bed, causing potential bed warping, but that is perhaps a separate/future issue/discussion, as a small bit of it is actually desirable to get the part to release.