MarlinFirmware/Marlin #25296: [BUG] G28 causes an out of bounce move on IDEX system when T1 is active

rondlh, 2023-01-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Issuing a G28 causes an out of bounce move when issued while T1 is active on an IDEX system. T1 tries to move to the homing position of T0, which is unreachable.

### Bug Timeline

Don't know

### Expected behavior

Move T1 within the predefined limits.

### Actual behavior

T1 tries to move to the T0 homing position, in my case it's stopped because of soft endstops.

### Steps to Reproduce

1. Issue a T1 on an IDEX system
2. Issue an G28 (home printer)
3. After the homing is completed, T1 tries to go to the home position of T0

### Version of Marlin Firmware

Latest (2023-01-23)

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### Add-ons

BTT TFT35

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

The configuration holds the settings for 3 printers, in this case IRON_IDEX is active.
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/10517009/Configuration_adv.zip)

toreandreas, 2023-01-29:
Confirmed. Do not use '#define Z_AFTER_HOMING  10'.
Use '//#define Z_AFTER_HOMING  10'.

closed 2023-05-02 by github-actions[bot]

nickm324, 2023-06-10:
This issue has existed since 2.0.X and still exists. This an AutoPark issue that T1 always tries to move to T0 home. This even occurs when doing a Filament Change, after the nozzle heats up T1 will quickly move left to T0 home and crashes into the T0.