MarlinFirmware/Marlin #25297: [BUG] SKR3 still reports TMC connection Error

mikeww1, 2023-01-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have just compiled a new firmware for my new SKR3 with TMC 2209 drivers. I killed my SKR2 board and had to replace it.
Despite of the various posts that the bug has been fixed by updating the environment, [https://github.com/MarlinFirmware/Marlin/issues/24703](url) the error still appears and I cannot move any axis.
I used 2.1.2 and the latest bugfix 2.1.X with same results. I checked stm32h7.ini but found it correct according to the mentioned post. 

### Bug Timeline

Maybe a new bug maybe it is the old issue

### Expected behavior

Printer to be able to use the drivers to move the axes.

### Actual behavior

No movement possible

### Steps to Reproduce

Set up the environment
Edit the config files
compile
store on the sd card and upload
the error appears

### Version of Marlin Firmware

2.1.2 and bugfix 2.1.X

### Printer model

Anet A8+

### Electronics

SKR3

### Add-ons

BLtouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10519876/Marlin.zip)

thisiskeithb, 2023-01-27:
> I killed my SKR2 board and had to replace it.

Are you sure you didn't fry your drivers as well? Providing an [`M122`](https://marlinfw.org/docs/gcode/M122.html) output would help.

Also, make sure the jumpers under your drivers are set to UART mode:
[image]

Your configs are not up to date/from the latest `bugfix-2.1.x`, so please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

thisiskeithb, 2023-01-27:
I just flashed the latest bugfix (67bc855) to an SKR 3 with five TMC2209s and there was no TMC CONNECTION ERROR.
<details><summary>Build details:</summary>
<p>

```prolog
Processing STM32H743Vx_btt (platform: ststm32@~15.4.1; board: marlin_STM32H743Vx; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32H743Vx.html
PLATFORM: ST STM32 (15.4.1) > STM32H743VI (1024k RAM. 2048k Flash)
HARDWARE: STM32H743VIT6 480MHz, 1MB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 @ 4.20200.221104 (2.2.0) 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TMCStepper @ 0.7.3
|   |-- SPI @ 1.0
|   |-- SoftwareSerial @ 1.0
|-- LiquidCrystal @ 1.5.1
|   |-- Wire @ 1.0
|-- SlowSoftI2CMaster @ 1.0.0
|-- SlowSoftWire @ 0.0.0+20220807061740
|   |-- SlowSoftI2CMaster @ 1.0.0
|-- IWatchdog @ 1.0.0
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- Wire @ 1.0
|-- Servo @ 1.1.2
Building in release mode
[snip]
```

</p>
</details>

mikeww1, 2023-01-28:
Thank you for your reply. It did not catch my mind that my drivers could be dead. I will test as you proposed and on monday new drivers are due to delivery.

mikeww1, 2023-01-30:
Ok the drivers were dead. With new drivers the error is gone. Thanks again for the hint !

closed 2023-01-30 by mikeww1