MarlinFirmware/Marlin #25302: [BUG] M600 leaves a gap on Ender-3 Pro with release-2.1.2, bugfix-2.1.x(02010200 and 02010300)

steinhelten, 2023-01-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After the M600 filament change, the printer leaves a gap in the part.  I get the same behavior whether I insert the M600 with Cura or generate the GCODE file by Cura without the M600 and then insert the M600 by hand.  In the attached files, the one that has _M600 as part of the file name is the one I changed manually.  I have tried this with release-2.1.2, bugfix-2.1.x configuration version 02010200 and bugfix-2.1.x configuration version 02010300.  I am reinserting the filament such that I get signifcant extrusion during the purge.  The PauseAtHeight script generated by the slicer shows the same behavior, for whatever that's worth. The GCODE file with the M600 inserted by CURA also has the Pause At Height script at layer 20 in addition to the M600 at layer 10.  The manually modified GCODE only has the M600.
[image]

### Expected behavior

To resume printing where it left off.

### Actual behavior

Leaves a gap after filament change

### Steps to Reproduce

Print an object with the M600 filament change, and either actually change the filament or simulate it by re-inserting the same filament.  Make sure you get some extrusion from the purge - if not, purge again or try re-inserting the filament and purge again.

### Version of Marlin Firmware

release-2.1.2, bugfix-2.1.x(02010200), bugfix-2.1.x(02010300)

### Printer model

Creatlity Ender 3 Pro

### Electronics

32 bit V4.2.7 STM32F103 RET6

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/10523506/configs.zip)
[GCODE.zip](https://github.com/MarlinFirmware/Marlin/files/10523526/GCODE.zip)

steinhelten, 2023-01-28:
I have a theory that I think would exactly explain what I'm seeing.  If the M600 retracts the filament before moving the nozzle from the parking location back to the object (after purging), and then doesn't advance it the retraction distance before restarting the print, I think that would cause exactly what I'm seeing, and would probably go unnoticed on most prints, which are much larger than my tiny test case, so so proportionally would be missing much less material.  How do I test my theory?

Johnny846, 2023-01-28:
I don't quite understand you issue (I don't know nothing about marlin), but to test your theory couldn't you use G92 (https://marlinfw.org/docs/gcode/G092.html) to set extruder position after filament change?

steinhelten, 2023-01-28:
You may have something there.  But I don't want to guess at the amount.  I'm going to try it with G11 ("recover", "un-retract") and see if that works.  Thanks for the idea!

steinhelten, 2023-01-29:
I watched it, and it sure looks like it pulls the filament back after the purge, but then I think it ignored my G11 because there wasn't a corresponding G10 first.  I'm going to try another way. I've looked up the PAUSE_PARK_RETRACT_LENGTH and that's how far I think I need to advance the filament.

Johnny846, 2023-01-29:
1. The Pause At Height script doesn't include M600? How do you change the filament in this case? 
2. In the normal g-code with only the manually added M600, try to add   G92 Exxxx.xx    after the M600 (Where   xxxx.xx   is the last extruder positon, in this case   E60.18732   ).

steinhelten, 2023-01-29:
Johnny846 - thanks so much for your help. I won't take everyone on the journey that I took, but with your help, I finally found a solution that works for me.  (Yes ,I did use the G92 after the M600 with some hand-calculated values that basically ended up doing what the below step accomplishes.)

So - I guess this maybe is not a bug.  I was able to get the behavior I needed by setting 
```
 #define ADVANCED_PAUSE_RESUME_PRIME          PAUSE_PARK_RETRACT_LENGTH  // (mm) Extra distance to prime nozzle after returning from park.
```
I will leave it to other, wiser, more experienced minds to determine whether having this as two separate values is a "bug" or feature...

steinhelten, 2023-01-29:
[image]
Forgot to include the proof!

closed 2023-01-29 by steinhelten

steinhelten, 2023-01-30:
It's pretty clear to me that, after the purge, the printer is retracting the filament before moving back to where it's going to start printing with the new filament, and unless I use the DEFINE I mentioned, it does not advance it again, so the first few moves to "print" have no filament.   My test for this was to use the G92 command you mentioned, set the extruder position to -5mm from where we left it (which indeed it is), then use a G1 with the same E value as the last print command before the filament change, and that worked.  It was only after that (and writing up the whole thing) that I saw this "PRIME" define and decided to try that.  I can also believe that on large prints, where 5mm is a small part of the first wall layer, it might go unnoticed, so many others may be experiencing this and not realize it.

steinhelten, 2023-01-30:
Oh, and as you can see from these posts, I'm not the only one to experience this:
https://www.reddit.com/r/3Dprinting/comments/sfty4z/m600_filament_change_resumes_printing_too_high/
https://github.com/MarlinFirmware/Marlin/issues/18943
and others...  I also believe that there are more people out there that are interpretting the wall gap as a Z position error, when it is actually an E position error.

Johnny846, 2023-01-30:
Regarding the Reddit thread, there seems to be many things presumably causing the same problem, and I do believe for some people it may be a problem in the z-axis (be it due to Marlin or hardware). 

"It's pretty clear to me that, after the purge, the printer is retracting the filament before moving back to where it's going to start printing with the new filament, and unless I use the DEFINE I mentioned, it does not advance it again, so the first few moves to "print" have no filament. " 
If the E-stepper actually retracts the filament after the filament change, and doesn't un-retract right when the print starts again, then it looks more like a bug to me (who again, doesn't understand much of Marlin). 
-In this case, if you can, it would perhaps be good to test an older version of Marlin that already has all the features/parameters for filament change/advance pause, and see if the issue still happens. 
-You could also test a configuration file configured with only the minimum required changes for it to work with your printer, leaving all other settings to their defaults, and see if the issue still happens. (I had a bed leveling bug/issue that would only happen when another completely different z-axis setting was changed). 
-I myself will also try to use advanced pause/filament change on my printer (Not a Creality Ender, mine is somewhat of a Prusa "Clone"), and if I have the same problem, I'll let you know. 

-You could perhaps also use M114 (https://marlinfw.org/docs/gcode/M114.html) to try and get E-stepper positions during the filament change procedure, but this seem more complicated and maybe would actually just make things more confusing.

Johnny846, 2023-01-30:
Sorry if it's late to ask, but what exactly did you mean with "The PauseAtHeight script generated by the slicer shows the same behavior, for whatever that's worth." in the original post? Does **just ** running the Cura-generated PauseAtHeigth script/g-code, without any filament change or M600 whatsoever during the PauseAtHeigth, also leave a gap in the part?

steinhelten, 2023-01-30:
It seems like a bug to me, too, but given that there's a workaround by possible misusing a define (or maybe this is what it's meant for), I've reached the point where I don't feel like putting a lot more time into it right now.  Maybe when I get a few projects done, I'll return to it, or if I hear from one of the people doing active development, that might motivate me, too.  When I watch the filament feed gears during the M600, it DEFINITELY retracts the filament before the move after the purge, and does not advance it again unless I define that PRIME symbol.

steinhelten, 2023-02-01:
An additional note on G11 - it wasn't just that there wasn't a preceding G10 - it turns out you have to define FWRETRACT for Marlin to even have these commands (which I don't).  But my G92/G1 E definitely worked.

dmarnel, 2023-03-11:
I just did a search to see if anyone else was having this exact same problem, and I found this thread. I also have been having this issue for quite a while now. I did a little debugging in pause.cpp, and in the resume_print() function, around line 680, I found this code (second line from the top) that adds a retract:

```
// If resume_position is negative
  if (resume_position.e < 0) unscaled_e_move(resume_position.e, feedRate_t(PAUSE_PARK_RETRACT_FEEDRATE));
  #ifdef ADVANCED_PAUSE_RESUME_PRIME
    if (ADVANCED_PAUSE_RESUME_PRIME != 0) {
      DEBUG_ECHOLNPGM("*** 4");
      unscaled_e_move(ADVANCED_PAUSE_RESUME_PRIME, feedRate_t(ADVANCED_PAUSE_PURGE_FEEDRATE));
    }
  #endif
```
If I comment out the second line shown above, I get perfect prints again. I'm assuming that resume_position.e is being set incorrectly if it is a negative value after loading and purging filament. There is even a comment above the resume_print() function that seems to validate this:

```
 *   -  FWRETRACT: Recover/prime from the prior G10.
 *   - !FWRETRACT: Retract by resume_position.e, if negative.
 *                 Not sure how this logic comes into use.

```