MarlinFirmware/Marlin #25315: Tronxy Marlin Version Crashes while Printing

DiekenFrost, 2023-01-29:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Tronxy Sends me this Configuration to Update my XE3 SE. Iv´d Configured my Board and the Printer Seems Working. After Several Minutes After Print Job Start my Display Resets and the Printer Restarts. I Can´t find the Fail. Can Anyone Help? Without Printjob everything is OK. I Only have to Adjust E Steps to 820.

### Bug Timeline

Tronxy Update

### Steps to Reproduce

I´ved Flashed 5 Times. Problem Shouldn´t go.

### Version of Marlin Firmware

2

### Printer model

Tronxy XS3 SE 

### Electronics

Stock V6 161121

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[F4xx-SIM480x320-firmware.zip](https://github.com/MarlinFirmware/Marlin/files/10531011/F4xx-SIM480x320-firmware.zip)

Sry Terminal don´t give any Information because Crash. Simply Disconnect

DiekenFrost, 2023-01-29:
The Zip Contains all Tronxy Printer.

ellensp, 2023-01-31:
This is not Marlin provided firmware
It comes from https://github.com/tronxy3d/F4xx-SIM480x320

This is a fork of Marlin, all support and issues for a fork of Marlin is provided by the  forks maintainers  tronxy3d

closed 2023-01-31 by ellensp