MarlinFirmware/Marlin #25316: [FR] Addressing four TMS2208 using a *same* serial port is not implemented

aguaviva, 2023-01-29:
### Is your feature request related to a problem? Please describe.

Right now for using 4 TMC2208 we need to use 4 serial ports. This requires using 8 pins.

We could have the 4 drivers use just one serial port if we were to use 'addressing', all we need is to use the same code the TMC2209 backend uses. 

For example, the following line should be enabled for the TMC2208: 
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/feature/tmc_util.cpp#L952
And along it all the related code.

It seems that the TMCStepper library would just require minor change, if someone could look into this I could look into the lib.

### Are you looking for hardware support?

No, the HW already has all the support needed

### Describe the feature you want

The same addressing working for TMC2209 should be implemented for the TMC2208.

EvilGremlin, 2023-01-29:
> we need to use 4 serial ports. This requires using 8 pins

no, just 4 pins

> the following line should be enabled for the TMC2208

now read 2208 datasheet and you'll se that it doesn't support addressing

closed 2023-01-29 by thisiskeithb

aguaviva, 2023-01-30:
You are right, thanks for clarifying. But 2.1.x still need 8 pins, 2 per driver. How comes you say it only needs 4?

EvilGremlin, 2023-01-30:
nope, one-pin UART is working for like 4 years now, you just set both rx and tx same

aguaviva, 2023-01-30:
You mean that I could set the pins for TX and RX to 40 and all should work?

https://github.com/MarlinFirmware/Marlin/blob/6d1a283d91606731cffe79d1afb1c8a2c7b427d8/Marlin/src/pins/ramps/pins_RAMPS.h#L351

I haven't found this in the documentation, where is it exactly?

Many thanks for the quick responses :)

aguaviva, 2023-01-30:
OK, in my case that I have 4 TMC2208 drivers, it works by using pins 63, 64, 65, 66. Each pin should be set as both the RX & TX for each of the driver UARTs.

ellensp, 2023-01-30:
One a mega2560, you must use a pin that has change interrupt feature.

Pins: 10, 11, 12, 13, 50, 51, 52, 53, 62, 63, 64, 65, 66, 67, 68, 69