MarlinFirmware/Marlin #25320: [BUG] 100k (#1) thermistors read too low at lower ambient temperatures (compared to 2.0.x)

lantren, 2023-01-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

On my BTT SKR mini e3 1.2 board both thermistors read between 12 and 8 degrees colder at room temperature when compared to 2.0.X

The only reason this is an issue for me is that frequently when the ambient temp is between 15-18 c the thermistors read below 5 c , preventing the printer from operating. 

I have swapped and replaced both thermistors , no change in behavior. 

### Bug Timeline

This is a new issue with 2.1.x

### Expected behavior

When ambient temp is ~ 16 c. The thermistor is expected to read > 5 c
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/10540225/Archive.zip)

### Actual behavior

Thermistor actually reads < 5 c. Preventing the printer from operating with 5 c low temp cut off. 

### Steps to Reproduce

Flash firmware to 2.0.x, observe reported temperatures @ ~ 16c ambient
Flash firmware to 2.1.x , observe reported temperatures @ ~ 16c ambient

### Version of Marlin Firmware

Marlin 2.1.2 

### Printer model

Creality Ender 3

### Electronics

BTT SKR Mini e3 1.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I've reviewed thermistor_1.h and there seems to be no change between 2.0.x and 2.1.x.

lantren, 2023-02-02:
It's strange that your efforts failed to identify the issue I'm experiencing. 

These are my actual readings with the current 2.1.x  @ ~ 15c ambient

Recv:  T:7.50 /0.00 (4006.00) B:5.50 /0.00 (4014.00) @:0 B@:0
Recv:  T:6.50 /0.00 (4010.00) B:3.75 /0.00 (4020.00) @:0 B@:0
Recv:  T:8.00 /0.00 (4004.00) B:4.37 /0.00 (4018.00) @:0 B@:0
Recv:  T:6.50 /0.00 (4010.00) B:6.50 /0.00 (4010.00) @:0 B@:0
Recv:  T:7.00 /0.00 (4008.00) B:5.50 /0.00 (4014.00) @:0 B@:0
Recv:  T:7.50 /0.00 (4006.00) B:7.50 /0.00 (4006.00) @:0 B@:0
Recv:  T:7.50 /0.00 (4006.00) B:4.37 /0.00 (4018.00) @:0 B@:0
Recv:  T:7.50 /0.00 (4006.00) B:2.50 /0.00 (4024.00) @:0 B@:0

Literally 45 seconds later,  after re-flashing to 2.0.x

Recv:  T:15.63 /0.00 (992.00) B:15.00 /0.00 (993.00) @:0 B@:0
Recv:  T:15.63 /0.00 (992.00) B:15.63 /0.00 (992.00) @:0 B@:0
Recv:  T:15.63 /0.00 (992.00) B:15.31 /0.00 (992.50) @:0 B@:0
Recv:  T:15.63 /0.00 (992.00) B:15.63 /0.00 (992.00) @:0 B@:0
Recv:  T:15.00 /0.00 (993.00) B:15.31 /0.00 (992.50) @:0 B@:0
Recv:  T:15.31 /0.00 (992.50) B:15.63 /0.00 (992.00) @:0 B@:0
Recv:  T:15.63 /0.00 (992.00) B:15.31 /0.00 (992.50) @:0 B@:0
Recv:  T:15.63 /0.00 (992.00) B:15.00 /0.00 (993.00) @:0 B@:0

This is with no other changes other than firmware. The thermistors and the same environmental conditions in both samples. 

It's also interesting to note the ADC values are reported differently. The reported values in 2.0.x are table equivalent.

lantren, 2023-02-02:
STM32F103RC_btt in both cases.

ellensp, 2023-02-02:
Could you do me a test...  

update motherboard to  #define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V2_0

build  2.1.x and attach the fimrware.bin, ill see if your build  failes on my hardware.

lantren, 2023-02-02:
Sure Thing 

for kicks I also included the terminal output.  
[Processing STM32F103RC_btt (board: genericSTM32F10.zip](https://github.com/MarlinFirmware/Marlin/files/10571599/Processing.STM32F103RC_btt.board.genericSTM32F10.zip)
[firmware.bin.zip](https://github.com/MarlinFirmware/Marlin/files/10571601/firmware.bin.zip)

ellensp, 2023-02-04:
Using your fimware.bin on my controller.
M115
> FIRMWARE_NAME:Marlin bugfix-2.1.x (Feb  2 2023 12:15:04) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

Confirming flash happened, 

M105
13:41:53.531 > ok T:9.56 /0.00 (3997.75) B:-15.00 /0.00 (4095.00) @:0 B@:0

Using my resistors set to 10c   (nothing plugged into the bed) 

I cannot replicate on my controller.

lantren, 2023-02-04:
Frag sebz zl vCubar

lantren, 2023-02-04:
Interesting. Then it must be something specific to the 1.2 board that doesn't occur on the 2.0 board

lantren, 2023-02-16:
If anyone else comes looking for a workaround for this or similar. 
Configuration_adv.h contains this in  /*** High Temperature Thermistor Support

//#define PREHEAT_TIME_HOTEND_MS 0
//#define PREHEAT_TIME_BED_MS 0

In my case an 10000ms bed preheat has been enough to raise the bed reading to over 5c in 15c ambient.

closed 2023-04-29 by github-actions[bot]

thisiskeithb, 2023-04-29:
To follow-up to this: I just flashed an SKR Mini E3 V1.2 with the latest `bugfix-2.0.x` (70c5bca) & `bugfix-2.1.x` (f9db5ab) and the bed & hotend thermistors show the same temps of 20C/21C in both versions of Marlin.