MarlinFirmware/Marlin #25323: [BUG] sudden overextrusion mid print

the3dfox, 2023-01-31:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

The problem I’m having is a sudden overextrusion which occurs after a random number of layers and will stay till the end of the print. When I start another print (with or without a reboot of the printer) the first couple of layers will be fine again until the sudden overextrusion starts again.
Reducing the flow to around 60% will let the printer extrude close to expected. Just as a side note. What I want to point out here is, that reducing the flow definitely has an impact on the extrusion amount. I also checked the steps/mm for the extruder as well as the flow rate reported by the firmware before and during the over extrusion and both values look to be fine (same before as during the problem). 
The Printer uses Marlin 2.1.1 with a TFT35v3 attached. The reason I think it is a marlin bug/problem and not a bug of the TFT firmware is because disconnecting the TFT and streaming the Gcode via Octoprint delivers the same results.

### Expected behavior

continuous extrusion throughout the print

### Actual behavior

sudden over extrusion mid print

### Version of Marlin Firmware

2.1.1

### Printer model

DIY

### Electronics

btt skr 1.4 turbo

### Add-ons

btt tft35v3

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/10547574/configs.zip)
[image]
[image]
[image]

Mjankor, 2023-01-31:
Just FYI, but I've seen something similar in my Z-axis. It would randomly double the steps on one of the motors. Looks like it was a dodgy cable and when disconnected the driver would reset and go from 32 microsteps back to 16. Might be worth replacing the stepper motor cable and seeing if that solves it.

the3dfox, 2023-02-01:
Congratulations, we have a winner. Exactly as you stated, the mircosteps changed from 16 to 8. Unfortunately replacing the cable didn't change anything. But I guess pinpointing the issue will be much easier from now on.
I will also try a different stepper motor just to be sure.

ellensp, 2023-02-02:
There have been a rash of tmc2209s of late that randomly reboot. replacing the stepper driver seems to be the only fix

ellensp, 2023-02-02:
There have been a rash of tmc2209s of late that randomly reboot. replacing the stepper driver seems to be the only fix.
One case on Y was a shoring Y end stop cable.  Anything connected to your  E0DET plug?

The-EG, 2023-02-02:
Since your board allows it, you can also swap the drivers around, I'd recommend E<->Z so you arne't crashing x/y if it happens again. If it follows the driver you may have one of the faulty ones. If not, you may have some other issue that can be addressed.

the3dfox, 2023-02-14:
Thank you all for commenting and hinting me in the right direction. It wanted to take my time to confirm the solution and before writing this answer. So far I had no issues anymore, and the solution was to change the pin for the filament sensor from E0 to E1. I did not try to remove the diag pin from the E0 driver, but I guess the result would be the same. I just found it cleaner to change the pin in firmware than to cut the diag pin off.

again thank you to all and I will close this issue now

closed 2023-02-14 by the3dfox