MarlinFirmware/Marlin #25324: [BUG] 'class Temperature' has no member named 'extrude_min_temp'

ReneHezser, 2023-01-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Allowing cold extrusion, will result in a build error.

```
//#define PREVENT_COLD_EXTRUSION
#define EXTRUDE_MINTEMP 170
```

Causes:
```Marlin\src\gcode\lcd\M145.cpp:50:72: error: 'class Temperature' has no member named 'extrude_min_temp'```

This has been a bug previously, but been fixed: https://github.com/MarlinFirmware/Marlin/issues/20194

### Bug Timeline

Old, but resolved as of issue #20194

### Expected behavior

I can compile 2.1.x (not bugfix)

### Version of Marlin Firmware

2.1.x

### Printer model

Self built

### Electronics

BTT SKR V1.4 Turbo

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-01-31:
This was fixed in bugfix last week: https://github.com/MarlinFirmware/Marlin/pull/25252, so pull down a fresh copy.

closed 2023-01-31 by thisiskeithb

ReneHezser, 2023-01-31:
Thanks a lot.