MarlinFirmware/Marlin #25325: MPCTEMP M306(E1 or E2) T,  reverts to E0

SlickNickeL, 2023-01-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

M306 E1 T or M306 E2 T reverts to only E0.  If a T1 or a T2 is first entered and then a M306 T is entered, it eventually reverts to previously selected active hotend (EO busy then shows E1 or E2)

### Bug Timeline

New

### Expected behavior

M306 E1 T should make hotend E1 active and test.
M306 E2 T should make hotend E2 active and test

M306 should operate as described in M306 Marlin online documentation. 

### Actual behavior

All M306 E1 or E2 T, reverts to only E0 as the active and tested hotend

### Version of Marlin Firmware

Bugfix-2.1.x

### Printer model

Hybrid multi head made for manufacturing 

### Electronics

Rumba

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-02-01:
This seem to be the same as https://github.com/MarlinFirmware/Marlin/issues/24962

SlickNickeL, 2023-02-01:
Yes and no.  Yes it seems that none of the parameters don't work. But at least the T1 T2 does tune the head, so it is usable but with the work around. As far as the other issue? I don't know. 
The tuning "once established through active head T T1 T2" did a nice job and outperforms PID especially on startups and first layer height startups. This is for an industrial manufacturing, 3 head printer, where time and repeatability, with 100% accuracy is a standard.

Bob-the-Kuhn, 2023-02-01:
Definitely a different issue.

Re-reading the M306 code I just realized that it says **T     Autotune the active extruder**.  Apparently M306 T was never intended to switch extruders.

Maybe we should change this to a feature request.

@ellensp - Any suggestions as to how to implement this feature?  @SlickNickeL and I did some work on this over on the [reprap Marlin forum](https://reprap.org/forum/read.php?415,891838).  I tried injecting a T command but that didn't work.

GMagician, 2023-02-01:
Remarks on m306 says about E as extruder, but it's not really used. m303 uses E as well then a proper choice could fall there

GMagician, 2023-02-01:
or better E is just used to set values and may be recycled to specify tune E

thisiskeithb, 2023-02-02:
Closing since there’s now a PR to address this: 
- https://github.com/MarlinFirmware/Marlin/pull/25326

closed 2023-02-02 by thisiskeithb