MarlinFirmware/Marlin #25328: [BUG] (bug summary) Neopixel crashes SD Card

jonpark08, 2023-02-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

### Basic Information

Printer: Kingroon KP3S
Board: MKS Robin Nano V1.2
MCU: GD32F303 (STM32F103 Alternative)
Firmware Version: Bugfix 2.1.X (Latest)

### Problem
When Neopixel Enabled, GCode files in SD Card are not able to be loaded. When connected to pronterface, the printer sends Error Messages(Image in the below).

### Expected behavior

Proceed to bed heating.

### Actual behavior

When Connected to Pronterface, Printer keeps sending:

`Error:SD read error`
`[ERROR] Error:SD read error`

### Steps to Reproduce

Enable `#define SDSUPPORT`

Enable `#define NEOPIXEL_LED`

### Version of Marlin Firmware

Marlin bugfix-2.1.x

### Printer model

Kingroon KP3S

### Electronics

BLTouch,  Neopixel Strip(11 LEDs)

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[image]

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10563978/Configuration.zip)

ellensp, 2023-02-02:
GD32 are not supported in Marlin.

jonpark08, 2023-02-02:
> GD32 are not supported in Marlin. And will not be supported until GD32 have been added properly to platformio

MKS Robin Nano V1.2 board has few variants because of the shortage of STM32 chipset.

It is possible to use GD32F303 by compiling it with STM32F103. Both have the same peripherals.

Neopixel works fine on Marlin 2.0.7 without no crash on SD Card.

By the way, I also will try to upload the firmware on STM32F103 version board.

ellensp, 2023-02-02:
gd32 chips are not fully compatible with stm32 chips. by design of the manufacturer. Where never designed to be drop in clones.

Here is a list of the better known differences 
[compatibility_sumup_between_GD32_and_STM32_V2.0.pdf](https://github.com/MarlinFirmware/Marlin/files/10564233/compatibility_sumup_between_GD32_and_STM32_V2.0.pdf)

jmz52, 2023-02-02:
@jonpark08, it is a known problem for GD32F303 MCUs and is unrelated to Neopixel.
https://github.com/MarlinFirmware/Marlin/pull/24980#issuecomment-1359589970

Problem is caused by flawed DMA controller in GD32F303 chip. 
GD32 DMA controller is not able to handle two concurrent data transfers - one for SDIO and another one for TFT screen.
STM32 has no problems with such configuration and code was tested on STM32 chips only.

As a temporary solution you can either switch to TFT_CLASSIC_UI or remove all SDIO defines and use SOFTWARE_SPI to access SD card.

jonpark08, 2023-02-02:
> @jonpark08, this is a known problem for GD32F303 MCUs and it is not related to Neopixel. [#24980 (comment)](https://github.com/MarlinFirmware/Marlin/pull/24980#issuecomment-1359589970)
> 
> Problem is caused by flawed DMA controller in GD32F303 chip. GD32 DMA controller is not able to handle two concurrent data transfers - one for SDIO and another one for TFT screen. STM32 has no problems with such configuration and code was tested on STM32 chips only.
> 
> As a temporary solution you can either switch to TFT_CLASSIC_UI or remove all SDIO defines and use SOFTWARE_SPI to access SD card.

Thanks for the reply.

ellensp, 2023-02-02:
closing as gd32 issue, needs real gd32 support.

closed 2023-02-02 by ellensp