MarlinFirmware/Marlin #25337: [BUG] Layer cooling fans only works with 100% of pwm speed (ATMEGA2560)

Henyckma-Sensei, 2023-02-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The layer cooler only works when the power percentage is at 100%, otherwise it does not activate, I use ramps 1.4 so I don't know if it happens on other boards
to make sure it was not an electrical problem, I went back to the previous version and everything worked perfectly.

### Bug Timeline

This is a new bug but have some related bug #25018 

### Expected behavior

I was expecting the printer to set the fan speed expressed by the slicer

### Actual behavior

The layer fan not are turning on while printing, doesn't do anything, I activate it at 75% in the menu and it doesn't turn on either... only if i turn at 100% can turn on but, only at full power.

### Steps to Reproduce

1.  I compiled and flash new marlin 
2. After flashing it, I started to print several objects
3. I realize that the layer fan doesn't work
4. I try to activate it from the lcd it doesn't work either (30%, 50%, 75%)
5. I increase it to the maximum speed and it works at full flow
6. flashed again with the old firmware
7. works normally, no problems

### Version of Marlin Firmware

2.1.2 and Bugfix

### Printer model

Custom prusa i3

### Electronics

Ramps 1.4 (EFB)

### Add-ons

none

### Bed Leveling

None

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10604037/Marlin.zip)

ellensp, 2023-02-04:
not bugfix configs...

Henyckma-Sensei, 2023-02-04:
> not bugfix configs...

> not current bugfix configs...
> 
> Updated to current bugfix for testing [bugfix 2.1.x Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10606707/bugfix.2.1.x.Configuration.zip)

thanks, I used the bugfix only to to see if the fan worked so i left everything blank and just changed the motherboard section

ellensp, 2023-02-04:
There is definitely an issue here, Oddly going back to 25018 where is worked find (there are pictures and 3rd parties verifying it) 

It still no longer works,,,   I am at a loss so far,

ellensp, 2023-02-04:
Issue seem to be the frequency, pwm is working.  

bugfix 2.1.x  62.5 khz
bugfix 2.0.x 980.4hz

ellensp, 2023-02-04:
I have a fix, 3 related bugs, works for fans.  Got to check it doesn't break pwming leds.. but that will have to wait..  its 4:30am.

Henyckma-Sensei, 2023-02-04:
> I have a fix, 3 related bugs, works for fans. Got to check it doesn't break pwming leds.. but that will have to wait.. its 4:30am.

I can wait as long as you want!, I appreciate your hard work, thank you!

ellensp, 2023-02-05:
@Henyckma-Sensei please test this PR

Henyckma-Sensei, 2023-02-05:
> @Henyckma-Sensei please test this PR

Works great!

ellensp, 2023-02-05:
Ok then, closing this as solved. PR will get merged into bugfix soon

closed 2023-02-05 by ellensp