MarlinFirmware/Marlin #25342: [BUG] Z Probing With Inverted Dual Z End Stops

Dimrain13, 2023-02-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When probing with the below settings on the SKR PRO V1.2 I get an error. "Error: Probing Failed measured_z : 0.0000" It then goes on to attempt to probe the next point. Initially I was using a touch plate and it had the same error using G29 P1 or attempting any ABL or UBL procedure so I attempted a BLTOUCH and it is giving the same error as the touch plate. I am running this as a CNC and it has dual Z endstops that home to max. I have attempted every Z_Probe setting under the sun and I come back to the same error. Whats odd is that it shows the current location on the line under the error. 

Currently I can use a probe to set the Z 0 point with G38.2 Z0 then G92 ZOffset, but this does not help with ABL or UBL functions.

The function order is as follows.
G28
G29 P1
Goes to the probe point
Z goes down
Probe hits and backs off
Probes again but slower
Error comes up
Attempts to probe the same spot until it gets to the final probe number
Goes to home position

Troubleshooting steps:
I have set the Z_PROBE_LOW_POINT to -200 as well in the hope that it would help, but no luck.
Changed Probe Offsets

### Expected behavior

G28
G29 P1
goes to first point
G29 P2
etc until finished.

OR

ABL option to completion

### Actual behavior

Probes one point failing each time then returns to home

### Steps to Reproduce

G29 P1 or ABL option in menu

### Version of Marlin Firmware

2.1.1

### Printer model

V1 CNC

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10609999/Configuration.zip)

Dimrain13, 2023-02-05:
Tested with Z_HOME Direction as the opposite as I need it to be and it works, but that is not viable for the setup that I have as a CNC.

Dimrain13, 2023-02-06:
Is there a check in place that I could disable that might be stopping the (G29) ABL/UBL functions from working? It seems like the probe is returning NAN looking at the code, but I do not see why. The Z_PROBE_LOW_POINT does not look to be triggered. One other item I noticed is that the Count is off for the Z axis, but Im not sure if thats relevant.

Dimrain13, 2023-02-07:
Set Z_MAX_POS from 200 to 0 and now auto bed leveling works. Issue seems to come from not knowing where zero is during the homing process. It is not a Z_HOME_DIR issue. Setting the Z_MAX_POS to 0 probes the wrong way with G38.2 Z0 so this is not a fix on its own. I will need to allow either untrusted home positions or something similar to have full required functionality.

Dimrain13, 2023-02-09:
I set Z_MAX_POS to 0 in the firmware default, after homing do G92 Z200 and no other settings changes and G29 works.
I set Z_MAX_POS to 200 in firmware, home, and G29 does not work. Only difference is that homing to 200 on the Z axis. To me that points to Z axis not being trusted. There is a check in motion.h I believe.

@thinkyhead

Dimrain13, 2023-02-10:
Almost to a work around. Setting Z_MAX_POS to 200 and then manual_Z_POS to 0 sets the actual Z position to trusted (Tried this in reverse last time), but then its  setting to 10 instead of the 200 after homing. Only thing I do not see a reason for it and when it tries to probe it wants to go up and hits the max z endstop but keeps going and then goes to do G29. When I set home it continues to show as Z10. I set #define Z_CLEARANCE_DEPLOY_PROBE  0 and it doesnt bottom out on it now. I have to manually do a G92 Z200 after homing to get proper functionality which is odd since I send Z_MX_POS to 200. 

Issues that are resolved:
Z Home DIR is up
G38.2 Probes down
G29 works

Possible solutions to avoid this:
Define Z_MAX_POS after trusting endstops for homing when homing to max for a given direction
Allow for homing to endstops as inherently trusted.

Dimrain13, 2023-02-20:
Anyone have a proposed solution based on the recommendations? I would think a simple Define Positive Z endstop that applies trust would do it.

thinkyhead, 2023-06-07:
Sorry that no one has replied to your query. We don't actually do this kind of support here. For best results getting help with configuration and troubleshooting, please use the following resources:

- Marlin Discord server. Join link: https://discord.gg/n5NJ59y
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-06-08 by thisiskeithb