MarlinFirmware/Marlin #25346: [BUG] EasyThreeD K7 STM32 (mks_robin_lite/mks_robin_lite_maple) - Firmware does not boot after Marlin 2.0.9.3

schmttc, 2023-02-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

- Compiled firmware for printer EasyThreeD K7 STM32 does not boot after Marlin 2.0.9.3 (ie. 2.0.9.3 works with _maple)
- Using default configuration.h and configuation_adv.h for board ET4000PLUS
- After updating to broken firmware, a connection fails using Repetier Host. Error: "Communication timeout - reset send buffer block". Printer does not respond to buttons or inputs.
- Using a working firmware, a connection is made and printer accepts commands.

Note: There are two board options for compiling:   mks_robin_lite_maple, and mks_robin_lite

**Test Matrix**
2.0.9.1 (pre-merged build) Only option for mks_robin_lite. This runs
2.0.9.3 (code accepted to main) mks_robin_lite_maple runs, mks_robin_lite does not run
2.0.9.4 mks_robin_lite does not run,  mks_robin_lite_maple does not run
2.0.9.5 mks_robin_lite_maple does not run, mks_robin_lite does not run
2.1 mks_robin_lite doesn't run, mks_robin_lite_maple doens't run

### Bug Timeline

Marlin 2.0.9.4

### Expected behavior

- A connection is made using Repetier Host, and command line responds to commands
- Printer responds to buttons and inputs

### Actual behavior

- A connection fails using Repetier Host. Error: "Communication timeout - reset send buffer block"
- Printer does not respond to buttons or inputs.
- 

### Steps to Reproduce

- Build firmware version after 2.0.9.3, using either mks_robin_lite or mks_robin_lite_maple as board
- Copy firmware to SD card
- Boot printer with SD card (triggers firmware update)
- Reboot printer and attempt to connect

### Version of Marlin Firmware

2.0.9.4

### Printer model

EasyThreeD K7

### Electronics

Board: ET4000+

### Bed Leveling

None

### Your Slicer

None

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-02-06:
Could you run a git bisect between the 2.0.9.3 and 2.0.9.4 using maple build env

Should be easy enough to find what broke it, later can look at non maple

Commits id's for these releases.

`f471eab1a2  (tag: refs/tags/2.0.9.3) 🔖 Marlin 2.0.9.3`
`369542db3b  (tag: refs/tags/2.0.9.4) 🔖 Marlin 2.0.9.4`

Please do the following

```
git clone https://github.com/MarlinFirmware/Marlin.git
git bisect start
git bisect good f471eab1a2
git bisect bad 369542db3b

```
It will reply with 
```
Bisecting: 208 revisions left to test after this (roughly 8 steps)
[7dae720de5c4fdea5fab657c09a2c8ba66f5a054] 📌 Fix, extend Index Rev03 Mobo (#23851)
```
{starts to repeate here:}           <---- just a label. you will loop back to here later.

Apply a minimal config and build with environment mks_robin_lite_maple and upload.
Test the build. does it work, ?

undo your config changes back to stock code. Ready for the next test

If it works enter
```
git bisect good
```
If it doesn't work enter
```
git bisect bad
```

git will select a new commit id to test and you go back to {starts to repeate here:}

At the end it will tell you what commit id broke it, and we can look more closely at what changed.

schmttc, 2023-02-06:
Thanks for the clear instructions. Output below:

b6abc760cef73a1dfc0e186f40687d0a45c8a769 is the first bad commit
commit b6abc760cef73a1dfc0e186f40687d0a45c8a769
Author: Scott Lahteine <github@thinkyhead.com>
Date:   Sat May 14 01:45:14 2022 -0500

    🔨 Move, tweak maple envs

:040000 040000 2cdf2b394dade40584b2f2d2632697de9c15ef79 8d580a4ef28d306e38e14a69d27fb87267240170 M      .github
:040000 040000 dfbf31b876a536ad77ae75bffa01f12520808176 8824c8658c39078ae79bca0a61201afd8af9a0a8 M      Marlin
:040000 040000 3d8e981a16f97adb94870e817c76d722baa662b0 a87c7b5dbf349034b6503970aca4bb3963c2bafd M      buildroot
:040000 040000 e2fb91056f5890495f9cbc8591a8188feb6f178c bd6fc9178352c87bc4b14f63463a2f96c6c4788c M      ini

ellensp, 2023-02-06:
One change I see that would effect your board is in ini/stm32f1-maple.ini

using latest bugfix 2.1.x

edit ini/stm32f1-maple.ini

replace
```
#
# MKS Robin Lite/Lite2 (STM32F103RCT6)
#
[env:mks_robin_lite_maple]
extends              = STM32F1_maple
board                = genericSTM32F103RC
board_build.address  = 0x08005000
board_build.rename   = mksLite.bin
board_build.ldscript = mks_robin_lite.ld
```

with

```
#
# MKS Robin Lite/Lite2 (STM32F103RCT6)
#
[env:mks_robin_lite_maple]
extends       = STM32F1_maple
board         = genericSTM32F103RC
extra_scripts = ${STM32F1_maple.extra_scripts}
  buildroot/share/PlatformIO/scripts/mks_robin_lite.py
```

and extract these scripts to buildroot/share/PlatformIO/scripts
[restore scripts.zip](https://github.com/MarlinFirmware/Marlin/files/10660213/restore.scripts.zip)

See if that is any better?

ellensp, 2023-02-06:
I was looking around the differences from the old and new binary's and noticed they where very different from 320 onward..

That number range alarm bells, some mks firmware are encrypted from 320 to 31040

A quick check back.

The original 2.0.9.3 code encrypted the firmware
When https://github.com/MarlinFirmware/Marlin/commit/b6abc760cef73a1dfc0e186f40687d0a45c8a769 was applied the encryption was lost.

edit ini/stm32f1-maple.ini

replace

```
#
# MKS Robin Lite/Lite2 (STM32F103RCT6)
#
[env:mks_robin_lite_maple]
extends              = STM32F1_maple
board                = genericSTM32F103RC
board_build.address  = 0x08005000
board_build.rename   = mksLite.bin
board_build.ldscript = mks_robin_lite.ld
```

with

```
#
# MKS Robin Lite/Lite2 (STM32F103RCT6)
#
[env:mks_robin_lite_maple]
extends              = STM32F1_maple
board                = genericSTM32F103RC
board_build.address  = 0x08005000
board_build.encrypt_mks   = mksLite.bin
board_build.ldscript = mks_robin_lite.ld
```

I suspect this one line change will fix it for maple.

schmttc, 2023-02-06:
Changing "board_build.encrypt_mks" in  ini/stm32f1-maple.ini was successful, the printer now boots with the latest firmware (there is one error in easythreed_ui.cpp relating to the use of "card.ls();", but I commented that out for testing).

Thank you for your fast and insightful assistance.

ellensp, 2023-02-07:
https://github.com/MarlinFirmware/Marlin/pull/24679 broke card.ls();   it added a compulsory flag, broke things...

ellensp, 2023-02-07:
Cannot see any obvious issue with mks_robin_lite

Try adding to the end of the mks_robin_lite environment

build_flags                 = ${stm32_variant.build_flags}
                              -DTIMER_TONE=TIM3 
                              -DTIMER_SERVO=TIM2

schmttc, 2023-02-08:
The DTIMER changes did not work for mks_robin_lite 

Here is the board schematic https://github.com/schmttc/EasyThreeD-K7-STM32/blob/Additional-Files/ET4000%2B.pdf

ellensp, 2023-02-09:
try 

```
#
# MKS Robin Lite/Lite2 (STM32F103RCT6)
#
[env:mks_robin_lite]
extends                     = stm32_variant
board                       = genericSTM32F103RC
board_build.variant         = MARLIN_F103Rx
board_build.encrypt_mks     = mksLite.bin
board_build.offset          = 0x5000
board_upload.offset_address = 0x08005000
build_unflags               = ${stm32_variant.build_unflags}
                              -DUSBCON -DUSBD_USE_CDC
```

schmttc, 2023-02-09:
No success for this either.

I note that in version 2.0.9.1 where the only board "mks_robin_lite", it is actually being built with _maple, and is named incorrectly in ini/stm32f1-maple.ini

thinkyhead, 2026-10-08:
This looks to have been addressed by #25349, merged since this report, and there's been no further activity here. Closing for now as part of 2.1.3 cleanup. If you still see the problem on current `bugfix-2.1.x`, please let us know and we'll reopen.

closed 2026-10-08 by thinkyhead