MarlinFirmware/Marlin #25347: [BUG] Fan calculation macro uses defines that do not exist yet (wrong import order)

lukaskuzmiak, 2023-02-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Pull request #24873 moved the `CALC_FAN_SPEED` macro into `Marlin/src/inc/Conditionals_adv.h`, the macro however, makes use of `FAN_MIN_PWM`, `FAN_MAX_PWM` and `FAN_OFF_PWM` which are undefined by default. The values for these defines are "fixed" (if undefined) in `Conditionals_post.h` (lines 2707-2731). But `Conditionals_post.h` is only imported **after** `Conditionals_adv.h`.

Therefore, in the default case of undefined `FAN_MIN_PWM`, `FAN_MAX_PWM` or `FAN_OFF_PWM` the current macro for `CALC_FAN_SPEED` always falls into the "else" clause because MIN/MAX are undefined:

```c
#if FAN_MIN_PWM == 0 && FAN_MAX_PWM == 255
  #define CALC_FAN_SPEED(f) (f ?: FAN_OFF_PWM)
#else
  #define CALC_FAN_SPEED(f) (f ? map(f, 1, 255, FAN_MIN_PWM, FAN_MAX_PWM) : FAN_OFF_PWM)
#endif
```

---

I am not sure where would be a good place for the `CALC_FAN_SPEED`, I'll be happy to submit a pull-request if anyone has a suggestion.

### Version of Marlin Firmware

2.1.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

sjasonsmith, 2023-04-01:
Can you confirm whether this is still an issue? If so, please attach configuration files which can be used to reproduce this.

GMagician, 2023-04-15:
I did some test and it seems confirmed with latest bugfix

GMagician, 2023-04-15:
@sjasonsmith you may use any configuration you want, I used mine, just comment out FAN_MIN_PWM and FAN_MAX_PWM in your "Configuration_adv.h" file and check which 'CALC_FAN_SPEED' branch is compiled (I added a simple #error but intellisense already display it)

closed 2023-04-23 by thinkyhead